RobotLocomotion/drake #1939: discuss/resolve the way we handle external dependencies

RussTedrake, 2016-03-25:
I've started a google doc here:
https://docs.google.com/document/d/1RMDja0UcU1t5ASRSeu7kW2ZuWMb7u6mhswWPP4QT4xU/edit?usp=sharing

Please comment!  My goal is to flesh this out, make some decisions, and the move this into sphinx documentation.

patmarion, 2016-03-25:
Perhaps the google doc can also indicate whether a dependency is required or optional, and if optional, what additional functionality it provides?

RussTedrake, 2016-03-25:
Added required/optional column

david-german-tri, 2016-05-25:
Since the Google Docs margin note conversations have died inconclusively (after a lot of great data gathering), I'll try to restart this thread.  

@jwnimmer-tri pointed out that the success criteria for changes to dependency management are unclear, and I agree.  @RussTedrake, maybe you can tell us more about the decisions you want this thread to yield?

liangfok, 2016-05-25:
Suggested success criteria: Drake successfully compiles and runs 100% of the time assuming users follow the [system configuration requirements](http://drake.mit.edu/developers.html#supported-configurations) and [installation instructions](http://drake.mit.edu/installation.html) to a T.

liangfok, 2016-05-25:
Suggested secondary success criteria: 100% of users can compile and run Drake without installing any proprietary or code with viral licenses.

david-german-tri, 2016-06-01:
friendly ping to @RussTedrake re. https://github.com/RobotLocomotion/drake/issues/1939#issuecomment-221438529

jwnimmer-tri, 2016-07-29:
I wonder if we should just close this issue?

david-german-tri, 2017-04-13:
I believe we should close this issue, yes.  The relationship to external dependencies will be more explicit and clear in Bazel, and we've already agreed that the supported, CI-tested modes will be fewer: "open source" and "everything".

closed 2017-04-13 by david-german-tri