RobotLocomotion/drake #1940: Improve on the accuracy of the contact and collision engine

amcastro-tri, 2016-03-25:
I've been working with height maps in #1616 and in my last test that you can see [here](https://www.youtube.com/watch?v=BOGBV8s14GQ) I get very unphysical results when a ball collides with a height map in an angle. 

I therefore made a simpler test of a ball bouncing on an inclined plane at 45 degrees. An analytical solution is available in this case. No friction, no damping. Second order RK. No multi-contact. You can see an animation [here](https://www.youtube.com/watch?v=2T47WVrhBFk).

[image]

As you can see the physics is also wrong in this simple example with convex objects (using Drake's original implementation).

amcastro-tri, 2016-03-25:
This is my plan for the upcoming days:

1. Submit a pull request with the height map implementation for #1616 
2. Work on a solution to handle concave objects for #1616. Using Bullet's collision dispatcher is an option.
3. Analyze this new issue #1940 to understand what collision forces act on the sphere and why. Solve it.

amcastro-tri, 2016-03-25:
We talked this morning with @david-german-tri about #1933 which relates to this issue and #1616. We agreed on taking the following path:
1. @david-german-tri will work an a purely Drake solution to filtering collisions using [`RigidBody::collidesWith`](https://github.com/RobotLocomotion/drake/blob/master/drake/systems/plants/RigidBody.h#L65).
2. I'll work myself with Bullet's dispatcher to handle also concave shapes
3. Hopefully @david-german-tri will be able to expand on my work to use Bullet's filters in the future.

sherm1, 2016-03-25:
The sideways bounce looks like a coordinate frame mixup. One talk I could give while I'm out in Boston (Apr 6-12) is how to write frame-intensive code using notation that makes it (almost) impossible to make mistakes. Hint: if transform is named "transform" or "pose" or "world2link" it is probably going to be wrong!

amcastro-tri, 2016-03-25:
Yes, that's a good observation. I always wondered what a good notation for this kind of things would be in code. I really liked the notation I found in [this](https://www.cs.ubc.ca/grads/resources/thesis/Nov02/Michael_Cline.pdf) thesis by Bradley Cline and I use it since. But this notation really useful in paper does not translate well to code variable names. 
Looking forward to your talk!

PR #1975 (2016-04-05): BUG_FIX: Implements fixes on the collision coordinate transformation used to compute collision forces
This fixes #1940. 

The fix is within RigidBodySystem::dynamics where the transformation `R` that transforms from the world frame into the collision point frame was incorrectly computed:

1. The transformation `R` was not a rotation but a rotation plus a reflection. The problem was on the computation of tangent2 which resulted on a frame which was not right handed.
2. The transformation matrix `R` that transforms from the world frame to the collision point frame was implemented as the transpos [... cut]
closed 2016-04-05 by RussTedrake