RobotLocomotion/drake #1950: installation instructions reference missing requirements.txt

RussTedrake, 2016-03-26:
looks like we missed a file.  
```
sudo pip install -U -r requirements.txt
```
is in http://drake002.csail.mit.edu/drake/sphinx/ubuntu.html (and perhaps other platforms) but that file seems to have not made it into the repo.

jwnimmer-tri, 2016-03-26:
Without derailing this too much (yes, we should capture the currently-required steps)...

Doing a `sudo pip install ...` on Ubuntu is extremely gross.  If I am reading it right, that will install python files directly unto `/usr/lib`, outside of the debian packaging system!  Or is there some `requirements.txt` phrasing that puts the packages into $HOME instead (in which case, why use sudo)?

liangfok, 2016-03-26:
I wouldn't worry too much about using `pip` since it's a standard packaging system for software written in Python. If putting things in `/usr/lib` is undesirable, we can use the `--target <dir>` flag:

```
sudo pip install -U --target $DRAKE_HOME/build/lib/python2.7/site-packages -r requirements.txt
```

jwnimmer-tri, 2016-03-26:
Pip is great.  My complaint is about the command that the setup instructions currently specify, which will screw up your Ubuntu in a nearly-unrecoverable way, AFAICT.  In general, `sudo` for anything other than `apt` commands is a defect in installation instructions.

rdeits, 2016-03-26:
I suppose whether it's "gross" or not is subject to opinion, but `sudo pip install` is entirely standard practice for installing python packages systemwide on ubuntu, since the python packages available from the standard apt repositories tend to be pretty out-of-date. And any package that was `pip install`ed can also be `pip uninstall`ed, so it's generally not unrecoverable.

liangfok, 2016-03-27:
A valid concern that was raised numerous times in recent weeks is whether the inclusion of a particular dependency will result in problems when Drake is integrated within a larger system that has a different set of dependencies (or versions of the same dependencies). How can we maximize the probability that Drake's dependencies do not clash when integrated into a larger system? Requiring that certain dependencies be installed in a system-wide manner seems risky in terms of inviting dependency clashes.

Another concern with relying on system-wide dependencies managed by `apt` or `pip` is the fact that Drake will have less control over when they are upgraded or deprecated. For example, I faced this problem with `yaml` when switching from Ubuntu 12.04 (which uses version 0.3) to 14.04 (which uses version 0.5). Drake will need to "keep up" with these system-wide changes as they occur.

Not saying we shouldn't rely on system-wide dependencies. Just need to be conscious about the consequences of doing so.

patmarion, 2016-03-27:
What packages are in requirements.txt?  Was pip being used because the ubuntu provided versions were too old?

jhoare, 2016-03-28:
It seems to me like the concerns about using ``pip`` could be alleviated by creating a ``virtualenv``  ( http://docs.python-guide.org/en/latest/dev/virtualenvs/ )

The downside of this, of course, is that the developer would need to activate the virtual environment first before doing drake activities. Drake doesn't provide a ``setup.bash`` or anything that sets up the user's environment does it?

rdeits, 2016-03-28:
Virtualenv is definitely a good solution if we want to avoid polluting the global system. Drake does actually produce `build/config/pods_setup_all.sh`, which is populated by the `pods_install_bash_setup` Cmake macro. That might be a good place to activate a virtualenv.

RussTedrake, 2016-03-28:
We have provided a setup.sh in previous projects (including the drc) but have resisted so far in drake.

patmarion, 2016-03-28:
Ok, but before discussing virtualenv can someone post the python packages listed in requirements.txt?  What python dependencies does drake have?  I can only think of director and iris externals having python dependencies.

jamiesnape, 2016-03-28:
Per the installation instructions, `sphinx` and `cpplint`. Everything else is installed by package managers.

jwnimmer-tri, 2016-03-28:
I'll suggest checking out cpplint from github as part of the superbuild, instead of firing off pip install during one-time setup.  I believe it will be a more successful mechanism to keep everyone using the same version of the tool.

patmarion, 2016-03-28:
can you use the python-sphinx package provided by ubuntu, or is the version too old to support drake's docs?

http://packages.ubuntu.com/trusty/python-sphinx

jamiesnape, 2016-03-28:
Possibly, but all the other platforms will use pip, so depends if you like to be consistent.

jamiesnape, 2016-03-28:
@jwnimmer-tri Sure. This will get people up and running for now.

PR #1961 (2016-03-30): Fix documentation regarding pip and Ubuntu
Fixes #1950.
closed 2016-03-30 by RussTedrake