RobotLocomotion/drake #1957: Standardize on top-level namespace "drake"

david-german-tri, 2016-03-27:
https://google.github.io/styleguide/cppguide.html#Namespace_Names

For instance,

```
namespace DrakeCollision {
```

should become

```
namespace drake {
namespace collision {
```

amcastro-tri, 2016-03-28:
On that note. I think we have bigger problems. Like having DrakeShapes::Element (systems/plants/shapes/Element.h) and DrakeCollision::Element (systems/plants/collision/Element.h).

I don't think I am that fond of changing variable's/namespaces/methods names at this stage. I think at some point when we all understand why Drake was designed the way it was we'll meet and try to decide whether or not we need to redesign interfaces. 

I personally prefer moving forward with the code as it is right now. I also feel the urge to do a complete redesign but I also feel I don't quite understand yet how/why Drake was designed the way it is. Hopefully @RussTedrake's technical talks will help us on this regard.

liangfok, 2016-03-28:
I prefer getting the code base style guide compliant as quickly as possible. 

Is there anything in the style guide forbidding two identically named classes in different name spaces?

liangfok, 2016-03-31:
I think we should augment Drake's style guide to include the following:

> The name space of a unit test's fixture implementation should match the name space of the class being tested plus an inner-most anonymous namespace.

This is to prevent name clashes in case one unit tests needs to include multiple fixtures.

sherm1, 2016-03-31:
> The name space of a unit test's fixture implementation should match the name space of the class being tested plus an inner-most anonymous namespace. This is to prevent name clashes in case one unit tests needs to include multiple fixtures.

By "include multiple fixtures" do you mean "link with multiple fixtures"? Having anonymous namespaces in headers would cause trouble.

liangfok, 2016-03-31:
Yeah, I just realized the word "include" should instead be "link" against. Thanks for catching that.

jwnimmer-tri, 2016-03-31:
> Standardize on top-level namespace "drake".

I agree, but this is going to break our API for downstream projects, so let's do it ~all at once, with a nice big warning note in the changelog (#1882).  If we can think of other major mechanical API changes, perhaps we should lump them in too.

jhoare, 2016-07-20:
In addition to this, since its related but not exactly the same thing: I've noticed that there are both ``Drake`` and ``drake`` top-level namespaces used in various places. According to style guide, all ``Drake`` instances should become ``drake``

liangfok, 2016-07-20:
How concerned are we about smoothly transitioning Drake so as to minimize headache for downstream users? Is there a way to `#DRAKE_DEPRECATED(...)` a namespace?

RussTedrake, 2016-07-20:
i don’t *think* many mit projects will be hurt by this change.

david-german-tri, 2017-04-13:
We've fixed most of the namespaces, but there are still some conspicuous violations - most notably `RigidBodyTree`. Today, unlike March 2016, I think there's consensus that the style guide is the standard, and developers will therefore fix this over time without a tracking issue.  Closing.

closed 2017-04-13 by david-german-tri