RobotLocomotion/drake #1958: jenkins make/test output should match exactly what developers see when they run make/make test at the command line

RussTedrake, 2016-03-27:
there is some confusion currently about why jenkins sometimes fails differently than the developer's workflow.  this could be clarified by:
- putting the build server scripts into the main repo (i understand they currently live on a fork by @billhoffman, presumably because they are still changing often)
- possibly updating the developer makefiles to run additional tests that are running on the build servers.

jamiesnape, 2016-03-28:
People also need to update to the latest installation instructions.

jamiesnape, 2016-03-28:
All the same tests should be running. Build flags also should now be the same. The Jenkins nodes are provisioned exactly per the build instructions. Development machines may not be. While Drake still depends on environment variables, there may be small differences, as Jenkins does not inherit the environment from the local machine (by design), so we have to manually set everything. Hence, Drake should not depend on environment variables.

jamiesnape, 2016-03-28:
Per the title, I cannot make the output match exactly. It will be close depending on developer configuration, but exactly is impossible given how Jenkins runs builds.

billhoffman, 2016-03-28:
For reference we are using a ctest script to drive Jenkins/CDash, it can be found here: 
https://github.com/billhoffman/drake/tree/build-scriptshttps://github.com/billhoffman/drake/tree/build-scripts

In order to get things to show up correctly it will not match exactly what you would see on the command line.  It is run via a ctest -S script.  https://github.com/billhoffman/drake/blob/build-scripts/scripts/drake_jenkins_ctest.cmake.   However, build errors and test failures should all be real and most likely due to different build options or because it is running with a different compiler/OS.

jwnimmer-tri, 2016-03-28:
I am **deeply** suspicious of a CI that runs different commands than me, at least for the primary platforms' PR pass/fail build configuration.  Perhaps Drake developers should learn how to run those CTest recipes to invoke the tests, so that we will be certain to get identical results?  If that is so, I can volunteer to start learning that tool, and update the developers.rst to match.

billhoffman, 2016-03-28:
It does not make sense to do that as the script submits to the CDash dashboad at the end.  You could make stuff like that optional, but I would not want to maintain the scripts for that use case as well.  Jenkins also "talks" to ctest via environment varaibles.  That would be a painful interface for users.  Set a bunch of environment variables and run ctest -S buildscript.   At the end of the day it does a configure, build, test, submit.  CMake should keep things consistent with the user build.  I have not seen this as an issue for VTK, ITK, and a host of other projects at Kitware.

jwnimmer-tri, 2016-03-28:
Ok.  I think for now then, I will wait until Russ's request atop this issue happens, and then see where that takes us.

david-german-tri, 2016-11-04:
I think we have gotten as close to this as it's possible to get with CMake.  Any further improvement will come from #3129.

closed 2016-11-04 by david-german-tri