RobotLocomotion/drake #1963: Visual Studio solution doesn't include header files

sherm1, 2016-03-28:
I'm using a CMake-generated solution file in Visual Studio 2015 IDE. Doing "Find in Files" for the "Entire Solution" returns matches in `.cc` and `.cpp` files, but nothing in header files.

In other CMake-built projects I've worked on, the "Entire Solution" find includes both source and headers, so I think the Drake `CMakeLists.txt` is missing something.

billhoffman, 2016-03-29:
The .h files need to be added to the source lists of the targets.  They will be ignored by generators that do not use them.  Drake does not seem to be doing that. https://github.com/RobotLocomotion/drake/blob/master/drake/solvers/CMakeLists.txt#L36

sherm1, 2016-04-18:
@billhoffman, any progress on this? It is hard to get around in the IDE without it. Can header dependencies be generated automatically somehow? It doesn't seem maintainable to list the explicitly.

david-german-tri, 2016-10-06:
Windows is no longer supported.

closed 2016-10-06 by david-german-tri