RobotLocomotion/drake #1964: Some compile errors can only be reproduced with licensed software

ggould-tri, 2016-03-29:
Changes that cause compile errors in SNOPTWrapper.cpp or any of system/plants/constraints/test/*.cpp cannot not be detected by builds on a developer machine that lacks SNOPT and Matlab (both paid licenses).  The same problem will likely occur with other licensed software (including software like Gurobi that is free only for academic users).

The correct approach is probably to create stub implementations of those packages' APIs that throw exceptions (to let the compile and link phases run on all Drake compilation units), and then to detect that condition in an appropriate place (eg, MathematicalProgram.cpp).  In the case of SNOPT this means that the current NoSNOPT approach is not really correct.

liangfok, 2016-03-31:
I agree with this issue because I just encountered a bug in `drake/systems/controllers/InstantaneousQPController.cpp` that wasn't detected because I do not have Gurobi installed.

jwnimmer-tri, 2016-04-04:
@RussTedrake This was the issue that came up in the leads meeting today.  Having a smarter approach to compile-time stubbing out the externals would make developer builds more likely to highlight errors before hitting the CI builders.

RussTedrake, 2016-04-05:
Yes - I totally understand.  Just have to get the licenses.

jwnimmer-tri, 2016-04-05:
The proposal in the top post (to change how stubbing is implemented) would be a way to find errors at compile time, without buying licenses.  If practical (it depends on the breadth of the proprietary APIs), it would help support the wider community of Drake developers that don't have access to licenses.

jwnimmer-tri, 2016-05-14:
I just hit this on `testIKtraj` as well.  The test should still be compiled even without `snopt_c` being around; it just shouldn't be run.

jwnimmer-tri, 2016-06-27:
I just hit this again, with `snopt`-related compile errors.

sammy-tri, 2018-09-21:
Closing since the build infrastructure is completely different now.  If anyone wants bazel build related work on an issue similar to this, I think we can track it in a new issue.

closed 2018-09-21 by sammy-tri