RobotLocomotion/drake #1973: Add more best practices to developers.rst

jwnimmer-tri, 2016-03-30:
Many Drake developers have best practices that are not documented for others to use:
- [ ] Use ccache.
- [ ] Use CMAKE_BUILD_TYPE=Debug.
- [ ] Compile within drake/drake instead of at the superbuild level.
- [ ] `git clean -fdx` instead of `make clean`.

There are probably more that I haven't heard about yet.

We should document these in `drake/doc/developers.rst` so that everyone can benefit.

liangfok, 2016-04-01:
- [ ] Run [pimpmychangelog](https://github.com/pcreux/pimpmychangelog) after updating the change log.

liangfok, 2016-04-01:
- [ ] When issuing a PR that will impact an issue, use the magic words listed [here](https://help.github.com/articles/closing-issues-via-commit-messages/).

amcastro-tri, 2016-04-05:
This is great @jwnimmer-tri! 
I would add that when running clang-format/cpplint for the first time on a given file(s) the format changes from clang-format/cpplint should be in a separate PR so that the original PR is clean from a large number of white-space corrections.

jwnimmer-tri, 2016-04-05:
I agree.  It looks like someone already editing the checklist to reflect this (yay).

amcastro-tri, 2016-04-05:
Nice @liangfok!

jwnimmer-tri, 2016-12-20:
Everything in this list is either (1) done or (2) moved to a specific issue or (3) only applies when using CMake.  I will close this.  Folks should feel free to contribute helpful PRs or file new issues as appropriate, but I'm not sure the meta-issue is helping anything here.

closed 2016-12-20 by jwnimmer-tri