RobotLocomotion/drake #1974: How to use a real robot with the InstantaneousQPController

alexbuyval, 2016-03-31:
Hi

I am trying to use the Drake software for Darwin-OP robot. I have simulated it successfully ([video](https://youtu.be/JT_2Es8jSJg)). However, I don't see way how to connect the real Darwin with Drake.

First of all, I faced out that I have position-Based servos on Darwin. I have found #1593 issue, but I don't understand what the current state of the issue. Have you finished the new the rigid body dynamics engine in C++ already?

The second, I have got the special Matlab [toolbox](http://www.mathworks.com/matlabcentral/fileexchange/46785-darwinop-robot-simulink-block) to control Darwin from Matlab. Is there way to use the simulink block from this toolbox together with InstantaneousQPController? Maybe there are any examples for other real robots?

Best Regards,
Alex

aidgab, 2016-04-07:
+1. Interested in running Drake on Nao. Are there any updates on position-based servos?

RussTedrake, 2017-08-01:
we're bringing up more humanoids examples now in the new c++ drake.  please check back again soon.

closed 2017-08-01 by RussTedrake