RobotLocomotion/drake #1978: RigidBodySystem is using a discontinuous contact model

sherm1, 2016-03-31:
Currently, [RigidBodySystem](https://github.com/RobotLocomotion/drake/blob/master/drake/systems/plants/RigidBodySystem.cpp#L144) contains a built-in compliant contact model that is essentially 
`force = k x + d xdot` 
where x>0 is penetration depth, k is stiffness, d is dissipation (damping) coefficient. That is a [Kelvin-Voigt material model](https://en.wikipedia.org/wiki/Kelvin%E2%80%93Voigt_material) which has the ugly property that it is violently discontinuous at initial contact, where x=eps and xdot>>0. Real materials can't generate dissipative forces until they are deforming. 

An equally simple, yet much better-behaved compliant contact model is [Hunt & Crossley's](http://appliedmechanics.asmedigitalcollection.asme.org/article.aspx?articleid=1402467), which looks essentially like 
`force = k x (1 + c xdot)`. 
Here the dissipative force is scaled by the compressive force, which is much more realistic, and the force is zero at initial contact regardless of xdot. Note that c here has units of 1/velocity and is not the same thing as d in the previous formula. c is related to the coefficient of restitution in an impact.

For a default compliant contact model, we should use Hunt & Crossley rather than Kelvin-Voigt since it is at least qualitatively realistic!

P.S. Depending on contact geometry the contact force may not be linear in x. For Hertz contact (like sphere-sphere), the Hunt & Crossley model looks like this: 
`force = k x^(3/2) (1 + c xdot)`.

/cc @amcastro-tri

RussTedrake, 2016-04-01:
+1 from me (the author of the initial model, not because of a strong stance but because it was the first thing out of my head when I was quickly bringing up that code).

SeanCurtis-TRI, 2016-10-13:
Somewhat related to this topic: reporting contact.

The mathematical definition requires x > 0.  This fundamentally excludes the magical case where two objects are osculating.  On one hand, that's the very physical definition of contact.  But in a proper model, it produces zero force.  So, when it comes to reporting contacts (see #3362), should this be excluded or classified as a zero-force contact?

sherm1, 2016-10-13:
Per f2f: we plan to report contact for x<=0 for compliant contact models but insist that the force is zero when x==0. (The current soon-to-be-dead Kelvin Voigt model can meet that requirement.) Note that this doesn't apply to active set models where "in contact" is a known state enforced by a constraint.

SeanCurtis-TRI, 2016-10-14:
Did you mean the "current soon-to-be-dead Kelvin Voigt model *can't* meet..."?

sherm1, 2016-10-14:
No, I thought you said that the K-V model actually did produce zero force at x==0, despite producing maximum dissipation at x==-eps. So by luck it still meets the zero-at-zero requirement. Is that wrong?

sherm1, 2018-08-14:
This was addressed in #5197.

closed 2018-08-14 by sherm1