RobotLocomotion/drake #1989: Simplest way to build in both debug and release modes?

aespielberg, 2016-04-02:
Coming from a CMake newb - is there a simple way to build in both debug and release mode simultanously?  If not, is there a relatively painelss way to switch it to Debug mode?  I tried changing the default target in the root Makefile to Debug and rerunning it after I had run it in release mode, but received many library runtime mismatch errors.  I'm assuming I need to change values in more places, is that accurate?

liangfok, 2016-04-02:
I'm not aware of a way to compile in both debug and release modes simultaneously. To compile in debug mode, I do not modify any makefile. Instead, I usually go into Drake's super-build level and execute:

```
BUILD_TYPE=Debug make
```

aespielberg, 2016-04-02:
Ah, you sniped my edit, bu tI'll operate temporarily under the assumption there isn't an error there.  super-build being /pod-build/drake-sueprbuild.sln?

liangfok, 2016-04-02:
By "super-build level" I'm referring to the same directory that contains the "drake" directory and other files like CONTRIBUTING.md, LICENSE.TXT, README.md, etc .

aespielberg, 2016-04-02:
Ah, I see, that's a command.  I'm assuming that's a *nix only way to do
it?  Such a command doesn't work on Windows.  Sorry, I should have
specified that I'm trying to get this working for Windows.

jwnimmer-tri, 2016-04-02:
There is a "make options" step in both:
 - http://drake002.csail.mit.edu/drake/sphinx/cygwin.html and
 - http://drake002.csail.mit.edu/drake/sphinx/windows.html

I believe that after you push `c` the first time there, you will see a new option `BUILD_TYPE` defaulted to `Release`.  Edit it and change to `Debug`, then to `c` again to configure, then `g` to generate what should be a debug build.

At least, that's how it works on Ubuntu and it seems like it should be the same for Windows.

aespielberg, 2016-04-02:
Hi Jeremy,

That is what I expected would happen.  However, on Windows, it pulls up the
CMake GUI.  The GUI has no such option.  I'm pasting a picture of all I see.

[image: Inline image 1]

jwnimmer-tri, 2016-04-02:
Wow, sorry this so is difficult.  Hopefully a Drake windows developer can help.  I'm useless, having only tested on Ubuntu so far myself.

aespielberg, 2016-04-02:
I am snot sure what the correct way to do things if you want build and release is.  I played around a little more with the bulid stuff.  Changing the root Makefile to have: 

```
ifeq "$(BUILD_TYPE)" ""
  BUILD_TYPE="Debug"
endif
```

which I believe is supposed to cascade to all the lower Makefiles below.

In this case, I get one linker error:

```
Build FAILED.

  "C:\ResearchCode\drake-debug\drake\drake\pod-build\ALL_BUILD.vcxproj" (default target) (1) ->
  "C:\ResearchCode\drake-debug\drake\drake\pod-build\systems\plants\IKoptionsmex.vcxproj" (default target) (5) ->
  "C:\ResearchCode\drake-debug\drake\drake\pod-build\systems\plants\collision\drakeCollision.vcxproj" (default target)
  (6) ->
  (Link target) ->
LINK : fatal error LNK1104: cannot open file 'BulletSoftBody.lib' [C:\ResearchCode\drake-debug\drake\drake\pod-build\sy
stems\plants\collision\drakeCollision.vcxproj] [C:\ResearchCode\drake-debug\drake\pod-build\drake.vcxproj]
```

which I believe is caused by an earlier error:

```
 Link:
    C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\bin\x86_amd64\link.exe /ERRORREPORT:QUEUE /OUT:"C:\ResearchC
  ode\drake-debug\drake\drake\pod-build\lib\Debug\drakeCollision.dll" /INCREMENTAL /NOLOGO /LIBPATH:"C:/ResearchCode/dr
  ake-debug/drake/drake/pod-build/lib" /LIBPATH:"C:/ResearchCode/drake-debug/drake/drake/pod-build/lib/Debug" /LIBPATH:
  "C:/ResearchCode/drake-debug/drake/build/lib" /LIBPATH:"C:/ResearchCode/drake-debug/drake/build/lib/Debug" kernel32.l
  ib user32.lib gdi32.lib winspool.lib shell32.lib ole32.lib oleaut32.lib uuid.lib comdlg32.lib advapi32.lib "-LIBPATH:
  C:/ResearchCode/drake-debug/drake/build/lib" BulletSoftBody.lib BulletDynamics.lib BulletCollision.lib LinearMath.lib
   ..\..\..\lib\Debug\drakeShapes.lib ..\..\..\lib\Debug\spruce.lib /MANIFEST /MANIFESTUAC:"level='asInvoker' uiAccess=
  'false'" /manifest:embed /DEBUG /PDB:"C:/ResearchCode/drake-debug/drake/drake/pod-build/lib/Debug/drakeCollision.pdb"
   /SUBSYSTEM:CONSOLE /TLBID:1 /DYNAMICBASE /NXCOMPAT /IMPLIB:"C:/ResearchCode/drake-debug/drake/drake/pod-build/lib/De
  bug/drakeCollision.lib" /MACHINE:X64  /machine:x64 /DLL drakeCollision.dir\Debug\DrakeCollision.obj
    drakeCollision.dir\Debug\Element.obj
    drakeCollision.dir\Debug\PointPair.obj
    drakeCollision.dir\Debug\Model.obj
    drakeCollision.dir\Debug\BulletModel.obj
    drakeCollision.dir\Debug\BulletResultCollector.obj
LINK : fatal error LNK1104: cannot open file 'BulletSoftBody.lib' [C:\ResearchCode\drake-debug\drake\drake\pod-build\sy
stems\plants\collision\drakeCollision.vcxproj] [C:\ResearchCode\drake-debug\drake\pod-build\drake.vcxproj]
```

The .sln doesn't generate in this case.  This bullet collision dependency is the same thing that fails if I try to do this alongside Release (although that case provides more errors).  Perhaps there is something wrong with the Windows build instructions with bullet, assuming this even is the correct workflow?

sherm1, 2016-04-02:
@aespielberg, if you are building with Visual Studio, CMake generates all the "Configurations" that are specified in the CMake variable `CMAKE_CONFIGURATION_TYPES`. Then you select the particular Configuration (Debug, Release, RelWithDebInfo) that you want to build while you are in Visual Studio. That variable is used instead of the `BUILD_TYPE` variable that is used on Linux.

XCode builds on Mac work the same way.

aespielberg, 2016-04-02:
Hi @sherm1 , thanks for the reply.  If that is the case, something else is wrong, because when trying to compile in Debug mode, I get very many errors.  Building ALL_BUILD in x64 Debug mode gives tons of errors, e.g.

```
Severity	Code	Description	Project	File	Line	Suppression State
Error	LNK2038	mismatch detected for 'RuntimeLibrary': value 'MD_DynamicRelease' doesn't match value 'MDd_DynamicDebug' in DrakeCollision.obj	drakeCollision	C:\ResearchCode\drake\drake\pod-build\systems\plants\collision\BulletCollision.lib(btConvexConcaveCollisionAlgorithm.obj)	1	
Error	LNK2038	mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in DrakeCollision.obj	drakeCollision	C:\ResearchCode\drake\drake\pod-build\systems\plants\collision\BulletCollision.lib(btDefaultCollisionConfiguration.obj)	1	
Error	LNK2038	mismatch detected for 'RuntimeLibrary': value 'MD_DynamicRelease' doesn't match value 'MDd_DynamicDebug' in DrakeCollision.obj	drakeCollision	C:\ResearchCode\drake\drake\pod-build\systems\plants\collision\BulletCollision.lib(btDefaultCollisionConfiguration.obj)	1	
Error	LNK2038	mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in DrakeCollision.obj	drakeCollision	C:\ResearchCode\drake\drake\pod-build\systems\plants\collision\BulletCollision.lib(btDbvtBroadphase.obj)	1	
Error	LNK2038	mismatch detected for 'RuntimeLibrary': value 'MD_DynamicRelease' doesn't match value 'MDd_DynamicDebug' in DrakeCollision.obj	drakeCollision	C:\ResearchCode\drake\drake\pod-build\systems\plants\collision\BulletCollision.lib(btDbvtBroadphase.obj)	1	
Error	LNK2038	mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in DrakeCollision.obj	drakeCollision	C:\ResearchCode\drake\drake\pod-build\systems\plants\collision\BulletCollision.lib(btCollisionObject.obj)	1	
Error	LNK2038	mismatch detected for 'RuntimeLibrary': value 'MD_DynamicRelease' doesn't match value 'MDd_DynamicDebug' in DrakeCollision.obj	drakeCollision	C:\ResearchCode\drake\drake\pod-build\systems\plants\collision\BulletCollision.lib(btCollisionObject.obj)	1	
Error	LNK2038	mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in DrakeCollision.obj	drakeCollision	C:\ResearchCode\drake\drake\pod-build\systems\plants\collision\BulletCollision.lib(btCollisionDispatcher.obj)	1	
Error	LNK2038	mismatch detected for 'RuntimeLibrary': value 'MD_DynamicRelease' doesn't match value 'MDd_DynamicDebug' in DrakeCollision.obj	drakeCollision	C:\ResearchCode\drake\drake\pod-build\systems\plants\collision\BulletCollision.lib(btCollisionDispatcher.obj)	1	
Error	LNK2038	mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in DrakeCollision.obj	drakeCollision	C:\ResearchCode\drake\drake\pod-build\systems\plants\collision\BulletCollision.lib(btCollisionWorld.obj)	1	
Error	LNK2038	mismatch detected for 'RuntimeLibrary': value 'MD_DynamicRelease' doesn't match value 'MDd_DynamicDebug' in DrakeCollision.obj	drakeCollision	C:\ResearchCode\drake\drake\pod-build\systems\plants\collision\BulletCollision.lib(btCollisionWorld.obj)	1	
Error	LNK2038	mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in DrakeCollision.obj	drakeCollision	C:\ResearchCode\drake\drake\pod-build\systems\plants\collision\BulletCollision.lib(btCollisionShape.obj)	1	
Error	LNK2038	mismatch detected for 'RuntimeLibrary': value 'MD_DynamicRelease' doesn't match value 'MDd_DynamicDebug' in DrakeCollision.obj	drakeCollision	C:\ResearchCode\drake\drake\pod-build\systems\plants\collision\BulletCollision.lib(btCollisionShape.obj)	1	
Error	LNK2038	mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in DrakeCollision.obj	drakeCollision	C:\ResearchCode\drake\drake\pod-build\systems\plants\collision\BulletCollision.lib(btConvexShape.obj)	1	
Error	LNK2038	mismatch detected for 'RuntimeLibrary': value 'MD_DynamicRelease' doesn't match value 'MDd_DynamicDebug' in DrakeCollision.obj	drakeCollision	C:\ResearchCode\drake\drake\pod-build\systems\plants\collision\BulletCollision.lib(btConvexShape.obj)	1	
Error	LNK2038	mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in DrakeCollision.obj	drakeCollision	C:\ResearchCode\drake\drake\pod-build\systems\plants\collision\BulletCollision.lib(btPolyhedralConvexShape.obj)	1	
Error	LNK2038	mismatch detected for 'RuntimeLibrary': value 'MD_DynamicRelease' doesn't match value 'MDd_DynamicDebug' in DrakeCollision.obj	drakeCollision	C:\ResearchCode\drake\drake\pod-build\systems\plants\collision\BulletCollision.lib(btPolyhedralConvexShape.obj)	1	
Error	LNK2038	mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in DrakeCollision.obj	drakeCollision	C:\ResearchCode\drake\drake\pod-build\systems\plants\collision\BulletCollision.lib(btConvexHullShape.obj)	1	
Error	LNK2038	mismatch detected for 'RuntimeLibrary': value 'MD_DynamicRelease' doesn't match value 'MDd_DynamicDebug' in DrakeCollision.obj	drakeCollision	C:\ResearchCode\drake\drake\pod-build\systems\plants\collision\BulletCollision.lib(btConvexHullShape.obj)	1	
Error	LNK2038	mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in DrakeCollision.obj	drakeCollision	C:\ResearchCode\drake\drake\pod-build\systems\plants\collision\BulletCollision.lib(btConvexConvexAlgorithm.obj)	1	
[... cut ...]
Error	LNK1318	Unexpected PDB error; ACCESS_DENIED (15) 'C:\ResearchCode\drake\drake\pod-build\systems\plants\dummy_mex.mexw64.pdb'	centerOfMassmex	C:\ResearchCode\drake\drake\pod-build\systems\plants\LINK	1	
Error	LNK1318	Unexpected PDB error; ACCESS_DENIED (15) 'C:\ResearchCode\drake\drake\pod-build\systems\plants\dummy_mex.mexw64.pdb'	centerOfMassJacobianmex	C:\ResearchCode\drake\drake\pod-build\systems\plants\LINK	1	
Error	LNK1318	Unexpected PDB error; ACCESS_DENIED (15) 'C:\ResearchCode\drake\drake\pod-build\systems\plants\dummy_mex.mexw64.pdb'	velocityToPositionDotMappingmex	C:\ResearchCode\drake\drake\pod-build\systems\plants\LINK	1	
```

sherm1, 2016-04-02:
Yikes! Looks like it didn't build a Debug version of Bullet.

aespielberg, 2016-04-02:
Yikes indeed!  Do you have any idea why that might be / a quick patch for
the interim?

-Andy S.

RussTedrake, 2016-04-02:
Just last night i merged a patch from kitware to the bullet pod to fix this (i think).  But we haven't update the drake ref yet.  Try changing the git hash for bullet in the drake-distro superbuild cmakelists.txy to master and see if that helps?

aespielberg, 2016-04-03:
This seems to work, but there's one hiccup.  I set BULID_TYPE to Debug in the Makefile, and doing this gave me the bullet .obj files I needed, and everything compiled.  But then it wouldn't compile for Release mode unless I ran make again with BUILD_TYPE Release (it didn't download the needed Release files, I believe).  After I do that, it won't compile for Debug mode.  It seems to be overwriting flags, and I'm not sure how to get both settings there at the same time (or if that's possible).  The errors look like:

```
Severity	Code	Description	Project	File	Line	Suppression State
Error	LNK2038	mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in DrakeCollision.obj	drakeCollision	C:\ResearchCode\drake\drake\pod-build\systems\plants\collision\BulletCollision.lib(btConvexPolyhedron.obj)	1	
Error	D8016	'/O2' and '/RTC1' command-line options are incompatible	cartesian2cylindricalmex	C:\ResearchCode\drake\drake\pod-build\util\cl	1	
Error	D8016	'/O2' and '/RTC1' command-line options are incompatible	ExpPlusPPTrajectoryEvalmex	C:\ResearchCode\drake\drake\pod-build\systems\trajectories\cl	1	
Error	D8016	'/O2' and '/RTC1' command-line options are incompatible	PPTmex	C:\ResearchCode\drake\drake\pod-build\systems\trajectories\cl	1	
Error	D8016	'/O2' and '/RTC1' command-line options are incompatible	barycentricInterpolation	C:\ResearchCode\drake\drake\pod-build\util\cl	1	
Error	D8016	'/O2' and '/RTC1' command-line options are incompatible	cylindrical2cartesianmex	C:\ResearchCode\drake\drake\pod-build\util\cl	1	
Error	D8016	'/O2' and '/RTC1' command-line options are incompatible	footstepCollocationConstraintsMex	C:\ResearchCode\drake\drake\pod-build\systems\robotInterfaces\cl	1	
Error	D8016	'/O2' and '/RTC1' command-line options are incompatible	lcmLogger	C:\ResearchCode\drake\drake\pod-build\util\cl	1	
Error	D8016	'/O2' and '/RTC1' command-line options are incompatible	lqrmex	C:\ResearchCode\drake\drake\pod-build\solvers\cl	1	
Error	D8016	'/O2' and '/RTC1' command-line options are incompatible	publishLCMLog	C:\ResearchCode\drake\drake\pod-build\util\cl	1	
Error	D8016	'/O2' and '/RTC1' command-line options are incompatible	realtime	C:\ResearchCode\drake\drake\pod-build\util\cl	1	
Error	D8016	'/O2' and '/RTC1' command-line options are incompatible	testQuatmex	C:\ResearchCode\drake\drake\pod-build\util\test\cl	1	
Error	D8016	'/O2' and '/RTC1' command-line options are incompatible	drakeMexUtil	C:\ResearchCode\drake\drake\pod-build\util\cl	1	
Error	LNK2038	mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in DrakeCollision.obj	drakeCollision	C:\ResearchCode\drake\drake\pod-build\systems\plants\collision\BulletCollision.lib(btDefaultCollisionConfiguration.obj)	1	
Error	LNK2038	mismatch detected for 'RuntimeLibrary': value 'MD_DynamicRelease' doesn't match value 'MDd_DynamicDebug' in DrakeCollision.obj	drakeCollision	C:\ResearchCode\drake\drake\pod-build\systems\plants\collision\BulletCollision.lib(btDefaultCollisionConfiguration.obj)	1	
Error	LNK2038	mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in DrakeCollision.obj	drakeCollision	C:\ResearchCode\drake\drake\pod-build\systems\plants\collision\BulletCollision.lib(btDbvtBroadphase.obj)	1	
Error	LNK2038	mismatch detected for 'RuntimeLibrary': value 'MD_DynamicRelease' doesn't match value 'MDd_DynamicDebug' in DrakeCollision.obj	drakeCollision	C:\ResearchCode\drake\drake\pod-build\systems\plants\collision\BulletCollision.lib(btDbvtBroadphase.obj)	1	
Error	LNK2038	mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in DrakeCollision.obj	drakeCollision	C:\ResearchCode\drake\drake\pod-build\systems\plants\collision\BulletCollision.lib(btCollisionObject.obj)	1	
Error	LNK2038	mismatch detected for 'RuntimeLibrary': value 'MD_DynamicRelease' doesn't match value 'MDd_DynamicDebug' in DrakeCollision.obj	drakeCollision	C:\ResearchCode\drake\drake\pod-build\systems\plants\collision\BulletCollision.lib(btCollisionObject.obj)	1	
Error	LNK2038	mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in DrakeCollision.obj	drakeCollision	C:\ResearchCode\drake\drake\pod-build\systems\plants\collision\BulletCollision.lib(btCollisionDispatcher.obj)	1	
Error	LNK2038	mismatch detected for 'RuntimeLibrary': value 'MD_DynamicRelease' doesn't match value 'MDd_DynamicDebug' in DrakeCollision.obj	drakeCollision	C:\ResearchCode\drake\drake\pod-build\systems\plants\collision\BulletCollision.lib(btCollisionDispatcher.obj)	1	
Error	LNK2038	mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in DrakeCollision.obj	drakeCollision	C:\ResearchCode\drake\drake\pod-build\systems\plants\collision\BulletCollision.lib(btCollisionWorld.obj)	1	
Error	LNK2038	mismatch detected for 'RuntimeLibrary': value 'MD_DynamicRelease' doesn't match value 'MDd_DynamicDebug' in DrakeCollision.obj	drakeCollision	C:\ResearchCode\drake\drake\pod-build\systems\plants\collision\BulletCollision.lib(btCollisionWorld.obj)	1	
Error	LNK2038	mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in DrakeCollision.obj	drakeCollision	C:\ResearchCode\drake\drake\pod-build\systems\plants\collision\BulletCollision.lib(btCollisionShape.obj)	1	
[... cut ...]
Error	D8016	'/O2' and '/RTC1' command-line options are incompatible	testSingleTimeKinCnstmex	C:\ResearchCode\drake\drake\pod-build\systems\plants\constraint\cl	1	
Error	D8016	'/O2' and '/RTC1' command-line options are incompatible	updatePtrRigidBodyConstraintmex	C:\ResearchCode\drake\drake\pod-build\systems\plants\constraint\cl	1	
Error	D8016	'/O2' and '/RTC1' command-line options are incompatible	testSingleTimeLinearPostureConstraintmex	C:\ResearchCode\drake\drake\pod-build\systems\plants\constraint\cl	1	
```

david-german-tri, 2016-10-06:
Windows is no longer supported.

closed 2016-10-06 by david-german-tri