RobotLocomotion/drake #1993: Design numerical accuracy tests for rigid body simulation

RussTedrake, 2016-04-03:
What are the right tests?  Don't need contact at first -- how about something as simple as an undamped pendulum (or multi-link pendulum) that preserves it's total energy?

sherm1, 2016-04-03:
First thought: I would partition these into three categories:
- tests of the rigid body model
- tests of numerical integrators
- tests of known-good integrators applied to known-good rigid body models

Error controlled integrators are notorious for covering up errors in rigid body models, especially when there are constraints involved. And symplectic integrators are awesome at conserving energy while producing terrible answers.

tri-ltyyu, 2018-11-09:
Per issue scrub, we have many tests for rigid body model, numerical integrator, etc. Closing this as tests will be done ongoing basis.

closed 2018-11-09 by tri-ltyyu