RobotLocomotion/drake #1994: Set -j for ctest call in test command of drake/Makefile

RussTedrake, 2016-04-03:
What's the platform independent way to figure out a good number of jobs?  I have some logic in my build scripts for finding the number of cores on each platform, but you must have a better way?

jwnimmer-tri, 2016-04-03:
I generally feel like build systems that decide on their own how many cores to use are mean-spirited. It should be something I control, not the build system.  I've always preferred having the user pass in their answer, and writing down a standard recipe in the sphinx docs (like `make test -j$(nproc)`).

Also, given the way test output gets interleaved under parallel make (not even at newline boundaries), you could even argue that the best default answer for most users is `-j1` only.

But maybe having a default in the build system is appropriate for this project, where users may not be expert programmers, not read the docs, or not know that `-j#` exists as a thing in make in the first place?  In which case I guess my minimal request is that if I specify a -`j#` myself on the command line, it overrides whatever default gets built into the `Makefile`.

RussTedrake, 2016-04-03:
good.  that would definitely be better.  so to rephrase the question:

right now calling make -j N test will not call ctest -j N.  how can we make that happen?

rpoyner-tri, 2016-04-04:
not a solution, but a workaround (linux/osx only):
`$ CTEST_PARALLEL_LEVEL=$(nproc) make test`

billhoffman, 2016-04-04:
Can we just expect people to call ctest directly?  Normal cmake work flow for me is make;  ctest -j N;  or ctest -R sometest.   I rarely use make test.

liangfok, 2016-04-04:
> Can we just expect people to call test directly?

I'm OK with that as long as it's documented on Drake's wiki. I see three places that need to be updated:

1. drake/doc/developers.rst
2. drake/doc/from_source.rst
3. drake/doc/ubuntu.rst

rpoyner-tri, 2016-04-04:
I see now that the 'make test' target is customized for dashboard uploads, which are probably not that useful to developers' own test runs. Maybe rename the target to clarify its intent, and document direct invocation of ctest?

jamiesnape, 2016-04-22:
I would just get rid of `make test` and rely on the tried and tested `ctest` functionality.

billhoffman, 2016-10-06:
CMake automatically adds the test target into makefiles it generates.  So, this will just be a documentation thing.  Seems like we can close this now.

jamiesnape, 2016-10-06:
This dates back to the hard coded Makefile, not a CMake generated one:
https://github.com/RobotLocomotion/drake/blob/master/drake/Makefile

closed 2016-10-11 by billhoffman