RobotLocomotion/drake #1996: Setting GCC default to 4.9

peteflorence, 2016-04-04:
Going through new Ubuntu build instructions, http://drake002.csail.mit.edu/drake/sphinx/ubuntu.html

Think we need a line that ensures switching default over to GCC 4.9, I'm getting message below,. `gcc -v` says I am still on 4.8.5

I'll look up a fix.  Also is the Sphinx documentation in git somewhere, and can I contribute to that directly?

peteflo@peteflo:~/drake-distro$ env CXX=g++-4.9 CC=gcc-4.9 make options
Configuring with CMAKE_FLAGS: -DCMAKE_BUILD_TYPE=Release
CMake Error at CMakeLists.txt:5 (message):
  GCC version must be at least 4.9

-- Configuring incomplete, errors occurred!
See also "/home/peteflo/drake-distro/pod-build/CMakeFiles/CMakeOutput.log".
make: *** [configure] Error 1

liangfok, 2016-04-04:
Yes, the Sphinx documentation is in [drake/doc](https://github.com/RobotLocomotion/drake/tree/master/drake/doc).

peteflorence, 2016-04-04:
Thanks @liangfok

jwnimmer-tri, 2016-04-04:
I believe the instructions are correct?  Did you install `g++-4.9-multilib`?  What does `g++-4.9 --version` report on the command line (without CMake or make involved)?

peteflorence, 2016-04-04:
Problem solved, thanks @patmarion . I had run `make` once prior to to going through platform-specific instructions.  `rm -rf pod-build` fixed my problem, it must have been cached on 4.8.5

closed 2016-04-04 by peteflorence

jwnimmer-tri, 2016-04-04:
Aha! Another vote for #1973's pending tip of:
 - _The build system isn't working? `git clean -fdx` from orbit, it's the only way to be sure_.