RobotLocomotion/drake #1998: get rid of perl dependency in matlab ctest wrapper

RussTedrake, 2016-04-04:
as discussed in #1992

billhoffman, 2016-04-05:
The FindMatlab in CMake has a matlab_add_unit_test that looks good and handles the same thing as the perl script here:
https://github.com/Kitware/CMake/blob/master/Modules/MatlabTestsRedirect.cmake

https://github.com/Kitware/CMake/blob/master/Modules/FindMatlab.cmake#L727

It would be good to use this, and we could copy the module into drake with some guards to remind us to remove it when we require a new enough cmake by default.  It would be good to contribute any changes back to CMake.  But, the code looks good from a read.

jamiesnape, 2016-04-22:
@billhoffman Can this be closed or is there more to do?

billhoffman, 2016-05-10:
This is working, but I am leaving the issue open because it still needs to be upstreamed in CMake.

billhoffman, 2016-10-06:
This has now been done.  Drake will use the new one when a 3.7 cmake is used on drake.

closed 2016-10-06 by billhoffman