RobotLocomotion/drake #2000: Magic "retest this please" is confusing; recommend "jenkins: retest this please" or similar.

jwnimmer-tri, 2016-04-04:
The magic "retest this please" github comment should have a clearer referent, something like "jenkins: retest this please" (or @drake-jenkins-bot by name, perhaps?).  Just about ~everyone who sees this message on their PR for the first time thinks that they have to do something.  ("I should retest it?  How do I do that?  I already tested it.")

I am not sure if we can just change sphinx to recommend the new spelling, or if the magic glue also has to change.

jamiesnape, 2016-04-04:
It is a non-trivial change, and our mileage may vary with the reliability of a non-standard phrase.

jwnimmer-tri, 2016-04-04:
In that case, maybe a better solution is just adding a clearer note to the new-developer training checklist (and making sure new developers actually review it).

jamiesnape, 2016-04-04:
So it turns out that text before and after the regex should be ignored. So we could try `jenkins test this please`, `jenkins test this please`, etc.,  without reconfiguration.

jamiesnape, 2016-04-04:
I seem to recall it not being 100% reliable, but it does not hurt to try.

jwnimmer-tri, 2016-04-04:
I tried it with the at-command form, and it jenkins dutifully started up: https://github.com/RobotLocomotion/drake/pull/1999#issuecomment-205535487

I will prepare a documentation PR changing the `jenkins.rst` to suggest this by default.

jamiesnape, 2016-04-22:
`@drake-jenkins-bot test this please` is working fine, so can this be closed? I see you have a follow up ticket #1973.

jwnimmer-tri, 2016-04-22:
Sure, it's just a doc problem now, so #1973 can cover it.

closed 2016-04-22 by jwnimmer-tri

tombroi, 2021-06-09:
retest this please