RobotLocomotion/drake #2004: Implement a solution to allow for concave shapes in Drake

amcastro-tri, 2016-04-05:
As mentioned in #1940, Drake currently does not support for concave shapes. Even when height maps (which in general are concave shapes) are introduced in #1944, Drake assumes convex shapes within [BulletModel::findClosestPointsBetweenElements](https://github.com/RobotLocomotion/drake/blob/master/drake/systems/plants/collision/bullet_model.cc#L426).

There could be two possible approaches to this:
1. Study how (and if possible) to extend [BulletModel::findClosestPointsBetweenElements](https://github.com/RobotLocomotion/drake/blob/master/drake/systems/plants/collision/bullet_model.cc#L426) to deal with concave shapes.
2. Directly use Bullet's dispatcher  and integrate it into Drake.

avalenzu, 2016-04-06:
+1 for #2 (using bullet's dispatcher).
BulletModel::findClosestPointsBetweenElements was designed for all-to-all
separation/penetration distance computations in the context of trajectory
optimization. It's been shoehorned into providing some general collision
support, but I think trying to make it the main routine for collision
detection in simulation is too much of a stretch.

amcastro-tri, 2016-05-25:
This is an update on the implementation of this dispatcher. I actually started off from the work by @avalenzu, mostly making the necessary changes so that the new solver can appropriately be integrated into the system dynamics. 

I tested this with the Cars example but I get a "shaky" car. Investigating the problem further it seems to be related on how Bullet reports manifolds. 
I decided to isolate a single wheel (or rather just a cylinder on a box) and start from there. 
See the [movie here](https://www.youtube.com/watch?v=jIRObowTtfc) showing collision points and forces on the cylinder.
From the video description:
> Bouncing cylinder on top of a box. Cylinder is shown as a wireframe so that collision points and forces are visible. 
Collision points are as computed by Bullet. Drake applies a "spring" force on each collision point. 
Notice that the collision points returned by Bullet are not symmetrical and therefore (since we have "forces-per-unit-point") the cylinder sinks more on its right side than on its left. 
In addition, as the number of collision points change, so does the resultant force on the cylinder by discrete amounts that ultimately lead to the instability of the solver.

@sherm1, @RussTedrake, @avalenzu: your input greatly appreciated.

RussTedrake, 2016-05-25:
looks like great progress.  

can you do a simple test where you move the tire slowly through a prescribed motion starting in the ground and coming out (at various angles) and see how stable the collision points reported by Bullet are?

amcastro-tri, 2016-05-25:
@RussTedrake, good idea. Is there a Drake-like way of prescribing motions? (I can do so by hand but I'd love to learn the Drake way).

sherm1, 2016-05-25:
Alejandro, that video looks great. I didn't see any obvious problem with Bullet's choice of contact manifold points for use with *rigid* contact (which is what Bullet thinks it's doing), but points that appear and disappear with non-zero penetration are a disaster for compliant contact since they violate conservation of energy. To model this way for compliance we will need to ensure that points are added and removed only when they are not penetrating. Maybe @erwincoumans has a suggestion for how that can be managed with Bullet? Otherwise we should be sure to include manifold stability as a feature of our new collision library. It would be interesting to see the behavior if the tire is modeled as a mesh rather than a cylinder.

RussTedrake, 2016-05-26:
Regarding how you would do the prescribed motion, maybe i overstated it.  I was just thinking you would manually sweep through some angles (calling doKinematics manually) and then call your collision detection code directly.  No need for the dynamics engine!

amcastro-tri, 2016-05-26:
Actually that is the first thing I did. However I would always get very deterministic collisions in contrast to what happens when I run dynamics. I did learn in the process that Bullet has "persistent manifolds" (btPersistentManifold) which seem to form a manifold with successive iterations through time. It seems to remember the last collisions and then it forms a manifold with a maximum of four points selected with some heuristics about how important these points will be for collision dynamics.
In my first tests (running doKinematics manually) I was just calling Bullet only once. But things get pretty strange as you call it in successive iterations.  
Still, the idea of calling doKinematics is a good one, with the caveat that we must call it several times to see something interesting.
However my hope is that for a first pass it will be good enough to return the maximum penetration depth point (which is what closestPointsAllToAll is doing).

erwincoumans, 2016-05-27:
You are implementing a custom collision detection implementation, just reusing some low-level parts of Bullet (right?) so it is hard for me to tell or predict what is happening. I'll still give some feedback that may help:

By default, using btGjkPairDetector/btGjkEpa2 and btPersistentManifold, only one single point is added each collision call, and points get updated and/or removed using some heuristic.

If you are using polyhedral convex hulls, there is also a different implementation in Bullet that computes all points at once, by clipping the polyhedral convex hulls against each-other. This is not enabled by default, but you can enable it with some steps: You would use btbtPolyhedralConvexShape::initializePolyhedralFeatures (btConvexHullShape is derived from this), and then enable SAT based clipping, using dispatchInfo.m_enableSatConvex = true. This is implemented somewhere in src/BulletCollision/CollisionDispatch/btConvexConvexAlgorithm.cpp as an alternative to GJK/EPA.

Even when using this single-shot contact generation, in Bullet, we would still use the btPersistentManifold, so we can store some extra information in contact points over time, such as warm-starting information of some LCP solvers.

Ideally you need a full boolean intersection operation (constructive solid geometry), computing an overlapping volume, and create a constraint that minimizes this volume.

Oh, and I'm still very interested in being involved in a new collision detection library, and unit tests. It will be more productive to discuss this in some meeting :)

sherm1, 2016-05-27:
Thanks, Erwin! I'm still trying to staff development of the new collision library. It will be great to have you involved once we get going.

sherm1, 2016-05-27:
Hey, @david-german-tri and @RussTedrake: the relevance of this discussion to System 2.0 might not be immediately apparent. But consider the "persistent manifold" mentioned by @amcastro-tri and @erwincoumans above. That is an example of a state variable of general, non-numerical type (that we've lumped into the "mode" variables at the moment). The previous constraint manifold (set of points) is saved in the mode variable. We ask the collision library to calculate a new set based on the old one, with the resulting new manifold used to calculate derivatives (and saved in the cache). Then if the step is accepted, the Solver uses the cached manifold to update the state variable.

The above operation can be done with no copying because the state variable and the cache update object are necessarily of the same type and can be swapped -- Simbody uses that approach.

david-german-tri, 2016-05-27:
@sherm1 Thanks very much for calling that out.  Your analysis makes sense to me, but non-numerical state in RigidBodySystem sounds like bad news.  I don't want to derail this thread, and the whole topic is probably easier to discuss f2f than on GitHub, but I've created #2467 as a bookmark.

amcastro-tri, 2016-05-27:
Thank you @erwincoumans! I will definitely give that a shot. Would you recommend modeling cylinders as polyhedral convex hulls as well? 

Right now I am calling `btCollisionWorld::performDiscreteCollisionDetection` and iterate through the points in each manifold and I just use the deepest penetration point in our dynamics. That works for simple geometries (tested with btConvexHullShape, btCylinderShape and btSphereShape).

Now I am setting up a `btBvhTriangleMeshShape` and it looks like many times collisions are missed and things just go through the triangle mesh (a static terrain). I need to investigate further if collisions are simply missed or returned with bad values, but I've seen people discussing this problem in other forums online.

Thoughts on this (new) problem?

erwincoumans, 2016-05-27:
For concave collision shapes, I recommend using convex decomposition (using HACD for example) and storing the convex parts in a btCompoundShape.

btBvhTriangleMeshShape is only suitable for static/non-moving geometries, such as buildings, ground etc and there is no algorithm for btBvhTriangleMeshShape versus btBvhTriangleMeshShape. So the question is: when you experience issues, what collision shape type is colliding against a btBvhTriangleMeshShape?

Ultimately, performing collision detection of a collision shape (say a btConvexHullShape) against a btBvhTriangleMeshShape is a combination of an acceleration structure (to cull triangles) and then pairwise convex-convex discrete overlap tests, between a single triangle and a btConvexHullShape.

Usually missing collisions ends up a user-side-issue. One reason to miss collisions is tunneling due to fast moving objects, which can be prevented using continuous collision detection (or approximations using a ray cast, sphere cast or convex cast etc).

If you can re-create a failing collision case in a simple Bullet example (for example modify the Bullet/examples/BasicExample.cpp) I can have a look at it.

amcastro-tri, 2016-05-31:
@erwincoumans: I am only using the btBvhTriangleMeshShape for a static terrain. Right now btBvhTriangleMeshShape is the only concave shape and all the other shapes are convex. 
I just found a bug introduced by myself but I am not sure this will solve all issues.
What I am doing right now is calling `btCollisionWorld::performDiscreteCollisionDetection` and then pick the point with the deepest penetration.
However, I realize now that this might not be the most recently computed collision point (probably the most stable but not the most accurate since when the collision objects start moving away I keep the deepest point for the next four time steps, the size of the cache). 
My question for you now is: is there a way to ask btPersistentManifold for the latest computed point? or to limit the number of cached points to just one?
Are there other kinds of manifolds in Bullet (like a manifold that only keeps the most up to date collision point).

erwincoumans, 2016-06-01:
The easiest way is to clear the entire contact manifold (remove all points), so the point you get is automatically the deepest/newest. No, there is currently no implementation of 1 contact point or other kind of manifold implementations. It was originally the intend to have various implementations, and I did experiment with various contact manifold ideas, but never merged those variations into main trunk.

>>Would you recommend modeling cylinders as polyhedral convex hulls as well?
This would be worth trying, in combination with enabling SAT to compute all contacts single-shot, instead of using GJK+EPA. You would need to call convex->initializePolyhedralFeatures() and enable m_enableSatConvex in the dispatcher info.

amcastro-tri, 2016-06-06:
@erwincoumans. I am testing with clearing the contact manifold. So far What I am doing is something like this:
```c++
for (int i = 0; i < numManifolds; i++) {
    btPersistentManifold *contactManifold = 
    bt_collision_world->getDispatcher()->getManifoldByIndexInternal(i);
    contactManifold->clearManifold();
  }
```
But I still l get results that seem to have some sort of coupling in "history". Is this the right way to clear manifolds?

I also tried using `btDispatcher::releaseManifold(btPersistentManifold* manifold)` but in that case I get a sef-fault some time after the call.

erwincoumans, 2016-06-14:
> I still l get results that seem to have some sort of coupling in "history".
> Is this the right way to clear manifolds?

Yes, btPersistentManifold::clearManifold is the right API. There should be no history, what kind of history? If this can be reproduced in a Bullet example/demo, I'll have a look.

amcastro-tri, 2016-06-17:
@erwincoumans: I am sorry but I do not have a Bullet-only demo. It all runs from withing Drake. I will try to explain:

We have a method (lets call it `MultiContact()` for the sake of this discussion) that calls `btCollisionWorld::setConvexConvexMultipointIterations` and `btCollisionWorld::setPlaneConvexMultipointIterations` on a Bullet world. 
In another method (lets call it `SingleContact()`), we do not make these calls.
Both methods use the same `btCollisionWorld` and call `performDiscreteCollisionDetection` on it. 

What I found out is that if I call `SingleContact()` before calling `MultiContact()` the results are not the same as if I had called `MultiContact()` without the previous call to `SingleContact()`. This is true even if I do clean the persistent manifolds.

We could of course have another bug somewhere else but that is the basic symptomatology I am observing.

amcastro-tri, 2016-06-17:
@erwincoumans: on a separate topic, I will give a try to the polyhedral convex hulls you mentioned before. As the name implies, does this functionality only work for convex geometries? Would it for instance work for `btBvhTriangleMeshShape`'s (I only use those for static terrains)?

Thank you!

PR #2596 (2016-06-28): Bullet collision dispatch from RBT
This PR builds on top of #2521 and #2543 to resolve #2004. 
It implements `RBT::ComputeMaximumDepthCollisionPoints` which dispatches collisions using [`Model::ComputeMaximumDepthCollisionPoints`](https://github.com/RobotLocomotion/drake/blob/master/drake/systems/plants/collision/Model.h#L84) and transforms collision points to bodies' frames as needed by [`RigidBodySystem::dynamics` here](https://github.com/RobotLocomotion/drake/blob/master/drake/systems/plants/RigidBodySystem.cpp#L143).

A Google test is implemented to test this capability.

---
This change i [... cut]
closed 2016-06-28 by sherm1