RobotLocomotion/drake #2011: Drake examples

beatsal, 2016-04-05:
I tried addpath_drake while in the drake directory but still get error; beed doing this for a day - no luck. Please help!

RussTedrake, 2016-04-06:
you'll have to tell us more than that for help.  what error are you seeing?

beatsal, 2016-04-06:
I am in c-users-admin-documents-matlab-drake-drake-Eigen-0.9.6-drake-examples-PlanarMonopodHopper. I key in:

>>cd PlanarMonopodHopper
>> StateMachineControl.run

I get the error:
The specified superclass 'HybridDrakeSystem' contains a parse error, cannot be found on MATLAB's search path, or is
shadowed by another file with the same name.

Kindly help.

Thanks

RussTedrake, 2016-04-06:
I agree that It looks like you have not successfully run addpath_drake.

beatsal, 2016-04-06:
I did add "addpath_drake" in the drake folder, so may be something wrong in drake?

RussTedrake, 2016-04-06:
what was the output of addpath_drake?

beatsal, 2016-04-06:
Yes, it was the addpath_drake command that was not working. I deleted and re-installed Drake 3 times and tried, the last time I put it in the Matlab folder and then only it worked. Maybe that should be in the install instructions or I missed it (more likely!).

I am really enjoying the course and thank you for putting together such a challenging course. The only thing missing is doing a useful project that makes use of the material learnt in the course. This could be fee-based.

Hope you will consider.

Sincerely,

Ken Dias

RussTedrake, 2016-04-07:
glad it works and that you're enjoying the course.

closed 2016-04-07 by RussTedrake

beatsal, 2016-04-26:
Sorry to bother you but don't know whom to ask. Think there is a bug in Drake:

Ex 3.6 of text book shows 2 cases which we are asked to try out. 'Quadrotor2D' works fine and a figure of the Quad is shown but not 'Quadrotor' (3D version) where no figure appears.

Thanks,

Ken