RobotLocomotion/drake #2015: Is Ubuntu 12.04 Support Still Needed?

david-german-tri, 2016-04-07:
There are various shards of documentation related to Ubuntu 12.04 in the wiki.  Open question: does the Drake community think it's important to continue 12.04 support?  Options are:

- Leave it in the current ambiguous state
- Officially cease support for 12.04 and remove references to it
- Officially support 12.04 with CI coverage
- Your ideas here

liangfok, 2016-04-07:
+1 for "Officially cease support for 12.04 and remove references to it"

Later this month: rapidly spin up official support for 16.04.

jamiesnape, 2016-04-07:
> Later this month: rapidly spin up official support for 16.04.

I think we should concentrate on getting every currently supported platform fully tested with all optional dependencies on CI first. Day one support of a new OS is not realistic at present.

patmarion, 2016-04-07:
The target platform for OpenHumanoids is Ubuntu 14.04, so with that requirement in mind I would not miss Ubuntu 12 support.

That said, I wish drake could support the default compiler on ubuntu 14... :)

david-german-tri, 2016-04-08:
Hearing no demand for Ubuntu 12, I'm going to drop it.  Thanks all!

PR #2039 (2016-04-11): Remove instructions for Ubuntu 12.04 Trusty, which is no longer supported.
Resolves #2015

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/2039) &emsp; Multiple assignees:&emsp;[image]&nbsp;<a href="/robotlocomotion/drake/pulls/assigned/RussTedrake">RussTedrake</a>,&emsp;[image]&nbsp;<a href="/robotlocomotion/drake/pulls/assigned/david-german-tri">david-german-tri</a>,&emsp;[image]&nbsp;<a href="/robotlocomotion/drake/pulls/assigned/jwnimmer-tri">jwnimmer-tri</a>
closed 2016-04-11 by david-german-tri