RobotLocomotion/drake #2016: gtest is causing (many) memory errors on the nightly dynamic analysis

RussTedrake, 2016-04-07:
https://drake-cdash.csail.mit.edu/viewTest.php?onlyfailed&buildid=5964

jwnimmer-tri, 2016-04-07:
The "use of uninitialized" is coming from a log statement during env-variable initialization.  Highly likely a false positive that should be suppressed.

The clang-msan can be very hard to debug unless you reproduce system libraries and build config exactly.  Is this builder a config we should be investing time on fixing?  If so, step 1 will be reproducing the error locally, while documenting in sphinx (or scripts) how to reproduce this build configuration.

david-german-tri, 2016-04-14:
Wow, that is a fascinating error.  I see no way the value (ptr_ in the scoped_ptr implementation in gtest-port.h) could actually be uninitialized, so my best guess is that we're linking against standard libraries that aren't instrumented for msan.  @jamiesnape, do you happen to know?

http://clang.llvm.org/docs/MemorySanitizer.html#handling-external-code

jamiesnape, 2016-04-14:
Looks like either a missing linker flag or one is not being passed down through the superbuild. It should not be a difficult fix and they are almost certainly false positives.

david-german-tri, 2016-04-15:
Nope, the flags are getting passed through just fine. From [jenkins 65](https://drake-jenkins.csail.mit.edu/job/linux-clang-nightly-memcheck-msan/65/consoleFull), you can see that googletest compiles and links with `fsanitize=memory` just like the other dependencies:

```
cd /home/ubuntu/workspace/linux-clang-experimental-memcheck-msan/pod-build/externals/googletest && /usr/local/bin/cmake -DCMAKE_INSTALL_PREFIX=/home/ubuntu/workspace/linux-clang-experimental-memcheck-msan/build -DCMAKE_BUILD_TYPE=Debug -DBUILD_SHARED_LIBS:BOOL= -DCMAKE_C_COMPILER=/usr/local/bin/clang -DCMAKE_CXX_COMPILER=/usr/local/bin/clang++ "-DCMAKE_C_FLAGS=-g -O1 -fno-omit-frame-pointer -fsanitize=memory -pedantic -Wall -Wextra" "-DCMAKE_CXX_FLAGS=-g -O1 -fno-omit-frame-pointer -fsanitize=memory -pedantic -Wall -Wextra" -DCMAKE_EXE_LINKER_FLAGS=-fsanitize=memory -DCMAKE_MODULE_LINKER_FLAGS= -DCMAKE_SHARED_LINKER_FLAGS=-fsanitize=memory -DCMAKE_STATIC_LINKER_FLAGS= -DCMAKE_VERBOSE_MAKEFILE=ON -DBUILD_SHARED_LIBS=ON -DCMAKE_INSTALL_NAME_DIR=/home/ubuntu/workspace/linux-clang-experimental-memcheck-msan/build/lib "-GUnix Makefiles" /home/ubuntu/workspace/linux-clang-experimental-memcheck-msan/externals/googletest
```

However, it appears we are not [linking against msan-enabled libc++](https://github.com/google/sanitizers/wiki/MemorySanitizerLibcxxHowTo), as I speculated above.  That explains why this build has been throwing lots of use-of-uninitialized-value failures [from the very beginning](https://drake-cdash.csail.mit.edu/index.php?project=Drake&filtercount=2&showfilters=1&filtercombine=and&field1=buildname&compare1=63&value1=msan&field2=buildstarttime&compare2=83&value2=2016-01-01).  googletest appears to predominate now only because it hits stdlib first in any test into which it's linked.

david-german-tri, 2016-04-15:
This would be nice to fix, but msan is a pretty low-priority build, given that we can run valgrind in reasonable time.  Let's put it on the Kitware backlog.

jamiesnape, 2016-04-15:
Sure. Though it is a literally a five minute fix.

david-german-tri, 2016-04-15:
Really?  In that case, go for it.  Thanks!

jamiesnape, 2016-04-15:
https://drake-cdash.csail.mit.edu/viewDynamicAnalysis.php?buildid=9629

jamiesnape, 2016-04-15:
`clang++ -stdlib=libc++ -L/usr/local/libcxx_msan/lib -lc++abi -I/usr/local/libcxx_msan/include -I/usr/local/libcxx_msan/include/c++/v1` etc.

closed 2016-04-15 by jamiesnape

david-german-tri, 2016-04-17:
Excellent, thanks!  Opened #2114 for the residual Bullet issues.