RobotLocomotion/drake #2030: Parameter Parsing in URDF/SDF

aespielberg, 2016-04-08:
This is a general discussion thread for how we should represent parameters in URDF/SDF.

It seems to me we have a few constraints:

-We need something that is backwards compatible (i.e. doesn't break other ROS-esque parsers).
-We need the expressiveness to represent arbitrary uses of parameters, e.g. where fields can be functions of parameters.  These are not necessarily "nice" in that they are not necessarily poly, e.g. inertias often look something like m/l^2, and it's possible m and l are both parameters.
-We need something simply parseable.  I think it would be possible to hook up something like GiNaC to do symbolic parsing and then substitute in parameter values or autodiffs, but I think many people are worried about C++ symbolic parser expressiveness and robustness.

My current recommendation for a temporary placeholder, is to, for any parameters, allow that fields only be direct assignments of parameters.  For example, mass = m1, or Ixx = I1, or fields like this.  Downstream, users can define in their application how these are constrained to one another, if at all.  This is not optimal at all and will have to change, since it's very cumbersome to, for instance, have to keep defining things which are scalar multiples of one another as different variables (this can happen a LOT).  These will be stored in new tags with suffix _sym, which will be ignored by other parsers.  The traditional non-sym tags then will be "defaults" which will be read in if the parser does not support parameterizations.

Constraints, whether explicit or implicit, will have to be supported in some way.  I'll continue to think about it, but wanted to just start this thread otherwise.

sherm1, 2016-04-08:
Adding some OSRF folks to this thread.

/cc @hsu, @nkoenig, @scpeters

scpeters, 2016-04-08:
We've had discussions in the past about how to represent parameters. I'll mention two issues: schema and espressiveness.

### Schema

A schema allows parameters to be described with types and validated on parsing. It is useful for known parameters, though it can be restrictive when experimenting with new parameters. We have discussed an approach for flexible parameter elements similar to `boost::any` in a [design document](https://bitbucket.org/osrf/gazebo_design/src/default/physics_params/physics_params.md).

### Expressiveness

Many urdf's in ROS use [xacro](http://wiki.ros.org/xacro) for parameterizing and repeating xml blocks. It also has support for simple arithmetic (`+-*/`). I've personally observed this to be very useful. The issue is that it's not a complete programming language. It starts with simple arithmetic, and then you get feature requests for if/else logic, trig functions, etc.

Another option is use to embedded scripting templates in xml, using ruby (erb) or python (empy). [Here is a model template](https://bitbucket.org/osrf/gazebo_models/src/d70c542675653b1787473df656c51517c6d47cea/cart_front_steer/model.rsdf?at=default&fileviewer=file-view-default) that expands to [a normal sdf file](https://bitbucket.org/osrf/gazebo_models/src/d70c542675653b1787473df656c51517c6d47cea/cart_front_steer/model.sdf?at=default&fileviewer=file-view-default). These models could be manually expanded, or ruby could be integrated into the sdf parser ([as was proposed before but rejected at the time](https://bitbucket.org/osrf/sdformat/pull-requests/118/adding-erb-parsing-to-sdf)).

We don't have any final answers, but this is some of what we've talked about before.

david-german-tri, 2016-04-08:
@liangfok FYI

aespielberg, 2016-04-08:
Simple arithmetic is not ideal, because, as you say, as soon as you allow simple arithmetic, people will want other functions.  Most notably, I think people would want trig functions, since describing revolute joints require them.

On that note, it might be worth allowing users to define metaparameters as a function of parameters in their own tags, and then allow for substitution later.

Embedding python is a cute idea, especially since Python is a core part of ROS.  Has there been more discussion on that particularly?

How in-demand have physical constants been?  Obviously, adding them at high preceision would be convenient so people don't have this tradeoff of typing in long numeric strings (unreadable) and short rounded versions (could lead to numerical error); as far as I understand this broke the Gazebo parser in the past but I'm not sure which characters in particular did so.

tri-ltyyu, 2019-03-22:
@EricCousineau-TRI, please see if this is still relevant and should be considered in current SDF work

EricCousineau-TRI, 2019-03-22:
Just to try summarizing my understand with `master` @ 767ec8868:
This is geared towards storing and parsing symbolic relationships among URDF / SDF entities with those files (e.g. getting `Parameters<Expression>`, `SystemConstraint<>`), and less so on embedding `xacro`-esque functionality within Drake's parsing.

This sounds like a great feature, but may not come any time soon yet, as it seems predicated on parsing directly from SDF -> `MultibodyPlant<Expression>` + `SceneGraph<Expression>`.

I've assigned it a low priority for now, and will update the title.

jwnimmer-tri, 2019-03-29:
I'd argue that having Drake load an SDF with placeholder names (and expressions) for real quantities, which the user binds to values later, is sufficiently far off our roadmap that we should just close this issue for now?  What possible value is this quiescent issue adding?

EricCousineau-TRI, 2019-03-29:
SGTM, esp. along the lines of "quiescent". We can re-open if there is an acute relevant use case.

closed 2019-03-29 by EricCousineau-TRI