RobotLocomotion/drake #2034: Adding Collision Models to Acrobot URDF Results in FunctionHandleConstraint Objects in Model

liangfok, 2016-04-08:
# Note: This issue is only relevant if https://github.com/RobotLocomotion/drake/pull/1898 is merged.

The problem with adding collision models is it results in unit test  [addRobotFromURDFStringTest.m](https://github.com/RobotLocomotion/drake/blob/master/drake/systems/plants/test/addRobotFromURDFStringTest.m) failing because the two `RigidBodyManipulator` models that are derived from the URDF cannot be compared using Matlab's [isequal()](http://www.mathworks.com/help/fixedpoint/ref/isequal.html) method. More details here: https://github.com/RobotLocomotion/drake/pull/1898#issuecomment-207197831.

As a temporary workaround, I created a version of Acrobot without the collision model called `Acrobot_no_collision.urdf` and modified addRobotFromURDFStringTest.m to use it rather than Acrobot.urdf, which has the collision models.

jwnimmer-tri, 2016-12-22:
Over to @sherm1 to decide if we care / reassign within his team / close?

sherm1, 2016-12-22:
Punting to @SeanCurtis-TRI. Sean, is there anything worth pursuing here? Should these Matlab tests be ported to C++ and nuked? Please feel free to close if this is no longer relevant.

SeanCurtis-TRI, 2016-12-28:
I'll take a look.

RussTedrake, 2016-12-29:
without looking at the details, it seems like a very reasonable action to take here would be to remove the `compareParsers` test.  We should just admit that the parsers are going to move forward in c++ and not in matlab (therefore the two will diverge).

jwnimmer-tri, 2017-06-19:
Closing as won't-fix, because #6369 will be removing the relevant software.  (Furthermore, I believe the test in question in this issue has _already_ been removed.)

If I'm confused and this should in fact stay open, please re-open an clarify the remaining unsolved problem(s) in a new comment here.

closed 2017-06-19 by jwnimmer-tri