RobotLocomotion/drake #2035: WinRM Hiccups Lead to Spurious CI Failures

david-german-tri, 2016-04-08:
e.g. https://drake-jenkins.csail.mit.edu/job/experimental/712/compiler=msvc-64,label=windows/console

We get 1+ of these per day.

jamiesnape, 2016-04-08:
Nothing I can do, unfortunately, short of filing bug reports with Microsoft.

david-german-tri, 2016-04-08:
Just for my edification, can you explain in a little more detail what's going wrong?

jamiesnape, 2016-04-08:
The WinRM connection restarts randomly for no reason. It is known problem and why you see the message `WinRM service responded. Waiting for WinRM service to stabilize....` for 3 minutes when the slave starts as an attempted workaround (it is more likely be drop earlier than later in the build).

jamiesnape, 2016-04-08:
The best you can do is limit your exposure. You may wish to consider the use cases for 32-bit builds.

jwnimmer-tri, 2016-04-08:
It is possible (easy?) to have a failure of results upload still leave the build green?  Perhaps make it a second build step whose result is ignored?

jamiesnape, 2016-04-08:
It would stay green, but it is not the upload that is failing. The slave is losing its connection.

david-german-tri, 2016-04-08:
Oh, interesting.  Is it just coincidence that the WinRM connection often seems to drop after `make test` succeeds?

jamiesnape, 2016-04-08:
Probably down to timing.

jwnimmer-tri, 2016-04-08:
Possibly results upload is saturating the link, causing WinRM to declare connection loss?

jwnimmer-tri, 2016-04-08:
Anyway, knowing nothing, perhaps I should bow out.  But +1 for any way that we can make merges not depend on winning the WinRM lottery.

jamiesnape, 2016-04-08:
I am assuming having stable builds is more important that uploading build artifacts for now?

jamiesnape, 2016-04-08:
The trouble is, currently build artifact upload is all platforms or none.

jwnimmer-tri, 2016-04-08:
The artifacts are... the compiled binaries? or just the build logs? or ...?

jamiesnape, 2016-04-08:
Both.

jamiesnape, 2016-04-08:
Probably more than we need, but code and logs are scattered everywhere.

billhoffman, 2016-04-08:
Since it is the default build that you get on Windows, I would say it should be supported.

david-german-tri, 2016-04-12:
Two questions: 

How hard would it be to disable build artifact upload on win32/win64?  If this is doable, I think we should try it just to see if it decreases flakiness.

How hard is it to upload build logs but not compiled binaries?  I can't really think of a reason we'd want to retrieve the compiled binaries from an experimental CI build (on any platform).

jamiesnape, 2016-04-12:
Artifact upload has been disabled since yesterday.

billhoffman, 2016-04-12:
I think it has been more stable.  I have not seen drops like we were getting.

jamiesnape, 2016-04-12:
After some research, apparently issues with large file uploads are also a known WinRM issue. Anything over ~2 MB is a problem. This is not limited to Jenkins, it is just that WinRM was not really designed with large uploads in mind.

jamiesnape, 2016-04-12:
When we start building nightly packages, we will need to work around this (not difficult, but something extra to configure).

RussTedrake, 2016-04-12:
I believe this must have been discussed, but i don't think we need the build artifacts from the PRs.  Just the cdash upload, which should be way below that threhold.  Correct?

jamiesnape, 2016-04-13:
Yes. There will only be artifacts on some nightly builds, not on PR or continuous.

jamiesnape, 2016-04-22:
Artifacts are off on Windows builds and they seem much more stable. Otherwise, short of rewriting the Jenkins plugin or WinRM client, there is not much more we can do for now, so this ticket is not really resolvable. Can this be closed?

closed 2016-04-22 by david-german-tri