RobotLocomotion/drake #2038: Is 32-bit Windows support still needed?

david-german-tri, 2016-04-08:
We currently have CI for both 32-bit and 64-bit Windows.  This is somewhat costly, and it increases our exposure to #2035 errors.  Open question for the community: is there a need to continue official support for 32-bit Windows?

sherm1, 2016-04-08:
Win32 :-1:

billhoffman, 2016-04-08:
It is the default build type for Visual Studio, so at least doing a nightly or a continuous might not be a bad idea.

RussTedrake, 2016-04-09:
This has come up many times now.  I originally added support due to a request from edx, but now think it's very valuable for catching alignment issues (which are real but could fail silently on different platforms).

david-german-tri, 2016-04-12:
Sounds like the conclusion is: Yes.  Closing.

closed 2016-04-12 by david-german-tri