RobotLocomotion/drake #2041: Should use consistent, error-resistant notation and conventions for spatial computations

sherm1, 2016-04-09:
Many errors in complicated 3D kinematics and dynamics code can be prevented by using consistent notation and conventions. Without that, code that mixes quantities in different frames is often developed by trial and error, leaving subtle bugs and low confidence. We have been finding and fixing such errors in Drake, but need a methodology that prevents them from happening and provides confidence in the code's correctness. Calling a rotation matrix `R` isn't enough, and a lack of test failures doesn't prove correctness.

A variety of nice notations are used in typeset papers, but few translate well to code. I use a particular notation with a long pedigree that can be mindlessly translated from math to code and back, has successfully produced amazingly complicated code that works right away, and can be verified by inspection and pattern matching. [Here](https://drive.google.com/a/trinst.com/file/d/0B-dn8WKPpzq4NnBTU184V2JlWHM/view?usp=sharing) are some slides about it.

I would like to move to a consistent, bulletproof notation for Drake and invite discussion and experimentation with this one.

sherm1, 2016-04-09:
Forgot to mention that the code examples are from Simbody; for Drake we would translate to the equivalent Eigen types in most cases.

rdeits, 2016-04-12:
@sherm1 I'd be interested in learning more about this, but I can't access those slides. Are they meant to be private?

sherm1, 2016-04-12:
Not private - I couldn't figure out how to get it shared outside TRI except
one email at a time. Hopefully I added yours now. Please let me know if
not. Regards, Sherm

sherm1, 2018-08-14:
I think we can declare this one done. (See Drake's [Multibody Concepts](http://drake.mit.edu/doxygen_cxx/group__multibody__concepts.html) documentation.)

Thanks everyone for adopting these conventions!

closed 2018-08-14 by sherm1