RobotLocomotion/drake #2043: Consider standardizing on double precision in Drake

sherm1, 2016-04-09:
Currently Drake contains lots of code with `VectorXd` and `Isometry3d`, explicit `double` variables, and constants of type double. For analysis, much of the code will need to be templatized by `ScalarType` where the meaning of `ScalarType` is defined by Eigen, and can include AutoDiff and anything else that satisfies Eigen's requirements, including just plain things like `double` and `float`. Part of Eigen's support for these scalars includes a requirement that there be a specialization of `NumTraits<ScalarType>`. That NumTraits specialization defines types `Scalar` (which will be float, double, complex<float>, etc.), and `Real` which will be float or double.

If we allow these ScalarTypes to have float precision, then we can't use all those types with explicit "d" at the end, even these are for objects that won't need ever to be AutoDiffed. Instead, every class will need to have code something like:
```c++
using Real = typename Eigen::NumTraits<ScalarType>::Real; 
using Isometry3 = Eigen::Isometry<Real, 3>; // roughly
// etc.
```
We could avoid this if we require the underlying precision of every ScalarType to use double, and we can enforce that using one `static_assert` at some high level.

RussTedrake, 2016-04-09:
not sure i understand.  there is almost certainly code that could use ScalarType instead of double in places.  that code could be inefficient if we set ScalarType=float, but it should still work.  There will be places where we'll get implicit float->double or double->float, or possibly compile errors, which would not be significantly different than your static assert.  And having the static assert means that all of our other types have to somehow declare their underlying precision -- I guess I feel like a symbolic variable should not have to.

sherm1, 2016-04-09:
I think we might be saying the same thing. I observe that the code has many variables in it that are explicitly double precision, like Isometry3d, VectorXd, double constants, etc. That to me seems inconsistent with the use of ScalarType, which could have any underlying precision (underlying the AutoDiff or polynomial or whatever). We should (a) remove those explicit precisions, or (b) require that the underlying precisions are always double.

RussTedrake, 2016-09-08:
@tkoolen might have an opinion

tkoolen, 2016-09-08:
My opinion is that we should remove those explicit precisions. 

The `Isometry3d`, `VectorXd`, etc. stuff is mostly used in older code written before we switched things to be autodiff/general scalar type compatible, and should be updated in the way described in the issue description.

Requiring the underlying precision to be double is in my view placing an arbitrary constraint that reduces functionality. It would also not work with `drake::FunctionalForm` in its current implementation.

sherm1, 2016-09-11:
Getting rid of explicit doubles is fine with me but I wonder if we are willing to do that for real (no pun intended!). That would mean, for example, that all those `1e-12` tolerances sprinkled throughout the code and test cases should instead be scalartype-dependent values. If we aren't willing to do that and verify that the code works in `float` precision (and I don't see what we would gain) then we ought to admit that we only support double when floating point computations are being performed, and enjoy the resulting code simplification.  

BTW, per [discussion](https://groups.google.com/a/tri.global/d/topic/simulation-archive/9L-A3OaHGh4/discussion) with @david-german-tri and @bradking, `FunctionalForm` ought to have a scalar type so that it can follow execution order as `AutoDiffScalar` does. Then it can be used as an empirical tool to explore functional dependence; currently that can only be done on non-branching code.

david-german-tri, 2016-09-12:
> BTW, per discussion with @david-german-tri and @bradking, `FunctionalForm` ought to have a scalar type so that it can follow execution order as `AutoDiffScalar` does. Then it can be used as an empirical tool to explore functional dependence; currently that can only be done on non-branching code.

Just to be clear, I think it makes sense to have both kinds of `FunctionalForm` - the current one that fails to compile in conditionals, and another backed by doubles.

> If we aren't willing to do that and verify that the code works in float precision (and I don't see what we would gain) then we ought to admit that we only support double when floating point computations are being performed, and enjoy the resulting code simplification.

I strongly agree that supporting 32-bit floats gains us nothing, and invites all manner of subtle bugs.  (Just try dividing by _c_!) However, I don't understand how we'd write a `static_assert` to ban 32-bit float, or how that would result in a code simplification per se.

sherm1, 2016-09-13:
> I don't understand how ... [banning] 32-bit float ... would result in a code simplification per se.

It wouldn't actually since our code is written as though it only needs to support doubles (hard coded tolerances like 1e-12, for example, and double constants in expressions). But if the code actually were written in a precision-independent way it would be more complicated. If we admit that we support only doubles, we can keep writing the code in this way. If we allegedly support floats also, then the code is defective. I'm saying we should formalize what we have already done informally (which I think is fine) and that is: support only double precision. That fixes all the defects with zero work.

david-german-tri, 2016-09-13:
Makes sense.  I agree.

jwnimmer-tri, 2016-09-21:
Compromise proposal:

(1) Document that our `T` (nee `ScalarType`) used throughout the codebase does not in general support single-precision floating point, though any function or class may choose to explicitly claim to (and be carefully tested in that configuration).

(2) Don't try to `static_assert` or otherwise fail-fast on `float` -- just let the docs be the answer.  (Leave it as "doctor, it hurts when I do this".)

(3) Possibly choose portions of the code where we should expect to (and test) supporting floats, e.g., `drake/math` primitives or conversion routines that touch other systems.  (For example, converting angle representations from floats that come in over LCM.)

sherm1, 2016-09-21:
I like that proposal. Is there an obvious place where that documentation would go?

jwnimmer-tri, 2016-09-21:
I think it's likely that doxygen for our `<T>` in every class where it means `ScalarType` should probably be citing to a master project-wide definition in Doxgen, which can then disavow single precision.

Or _shrug_ put it in Sphinx somewhere.  I don't think it'll matter much either way.

RussTedrake, 2016-11-27:
i think this once is safely resolved.  decision is that the systems classes should support only
`double`, `AutoDiffXd` (to be replaced asap by the chunked version), and `SymbolicExpression`.

closed 2016-11-27 by RussTedrake