RobotLocomotion/drake #2045: trajctoryOptimization location?

aespielberg, 2016-04-09:
Hi guys,

Been a while since I've used the MATLAB side, I've noticed some things have changed.

Trying to run 'runDirtran' in 'examples/KneedCompassGait.'

However, in running it, I get the following error:

```
No appropriate method, property, or field 'trajectoryOptimization' for class 'PlanarRigidBodyManipulator'.

Error in runDirtran>runDirtranIteration (line 161)
[utraj,xtraj,info] = p.trajectoryOptimization(@cost,@finalcost,x0,utraj0,con,options);

Error in runDirtran (line 4)
[xtraj,utraj,ltraj,v,p]=runDirtranIteration;
```

Has trajectoryOptimization been moved to be a member method of Manipulators?  If so, why isn't it working?

RussTedrake, 2016-04-09:
It doesn't invalidate the question, but note that that example is in the dev directory -- which means it was either not finished or not maintained.

aespielberg, 2016-04-09:
Good point.  Out of curiosity, do you know if it is on the roadmap to move in this direction?

RussTedrake, 2016-04-09:
Ah.  This is probably calling the super old trajectory optimization interface (which is probably why it got moved to dev).  The newer examples all use the DirectTrajectoryOptimization classes in the solvers directory.  That is the future roadmap, including implementing those classes in c++ too.

closed 2016-04-09 by RussTedrake