RobotLocomotion/drake #2046: Windows build still has some rough edges, especially for Debug builds

sherm1, 2016-04-09:
On Windows, Release or Debug is selectable from inside Visual Studio; it doesn't require a separate run of CMake. (I believe this is similar on OSX with XCode.) Our build system *almost* does this correctly, but switching to Debug in VS causes it to attempt to link Debug Drake code to Release Bullet code resulting in link errors. @MikeElkins-TRI and I determined that the superbuild has to be run with an environment variable `BUILD_TYPE` set to `Debug` in order to get the correct Bullet libraries built. There are a few other awkward pieces (see below).

@billhoffman, can this be made smoother?

From Mike's blog summarizing the strategy we followed successfully, with problems encountered:
```
* Do a top level build (super build) from the command line first.
* Open the solution file drake-distro/drake/pod-build/drake.sln, or other sub-project file
  if you prefer. It seems like they all end up opening everything anyway.
* The build system doesn't automatically set the PATH right to run without help. 
  Either change your global PATH to include drake-distro/build/lib, or right click in
  VS the project you want to run, click properties, and under Debugging->Environment, 
  add the text PATH=C:\Users\MYNAME\drake-distro\build\lib;%PATH%.
* Broken stuff persists in the cache to an undesirable extent. Before you waste
  TOO much time, blow everything away and rebuild. "git clean -dfx" means 
  "Nuke from orbit, it's the only way to be sure."
* Found a BUG! Visual Studio can only build debug builds of the Drake code if the 
  environment variable BUILD_TYPE is set to Debug.
```

aespielberg, 2016-04-09:
#1989

sherm1, 2016-04-10:
Here are a few more issues from my notes:
- After a complete “make”, going into visual studio and running anything requires a partial rebuild.
- This comes up whenever running in Debug in Visual Studio:
```
        This project is out of date: lcmtype_agg_hpp - Debug x64
```
- Here is the error message you get if you try to switch to the Debug configuration in VS 2015, if you didn't set `BUILD_TYPE=Debug` before doing the superbuild:
```
4>BulletCollision.lib(btDefaultCollisionConfiguration.obj) : error LNK2038: mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in DrakeCollision.obj
4>BulletCollision.lib(btDefaultCollisionConfiguration.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MD_DynamicRelease' doesn't match value 'MDd_DynamicDebug' in DrakeCollision.obj
4>BulletCollision.lib(btDbvtBroadphase.obj) : error LNK2038: mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in DrakeCollision.obj
4>BulletCollision.lib(btDbvtBroadphase.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MD_DynamicRelease' doesn't match value 'MDd_DynamicDebug' in DrakeCollision.obj
4>BulletCollision.lib(btCollisionObject.obj) : error LNK2038: mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in DrakeCollision.obj
```

david-german-tri, 2016-10-06:
Windows is no longer supported.

closed 2016-10-06 by david-german-tri