RobotLocomotion/drake #2047: Construct Visualizer silently fails on Windows

aespielberg, 2016-04-09:
plant.constructVisualizer() doesn't construct the visualizer or do anything meaningful.  I'm assuming this is because the BotVisualizer is created by Director, and according to #1786 , this is still in development.  Is there any interim visualization method that can be used in the meanwhile?

patmarion, 2016-04-09:
The Drake release for Windows release includes a working version of Director.  Launch Director manually using the supplied .bat script.

https://github.com/RobotLocomotion/drake/releases

aespielberg, 2016-04-09:
Sorry, I should specify this is from source.  Where is the .bat file?

patmarion, 2016-04-09:
The visualizer that is available in the zip is compatible with the source.  You can download the drake zip just for the visualizer, and use it with your source build.  The bat file is build/drake_visualizer.bat.

There are some mesh loading features that might not work in this version of the visualizer, since it is an old build, but it's worth a try.

closed 2016-04-09 by aespielberg