RobotLocomotion/drake #2058: Optimize The Way Floating Joints Are Added To A Robot Model

liangfok, 2016-04-11:
Currently, each time a new model is added to a rigid body tree, every link in the tree is searched regardless of whether it is a member of the newly added model. This occurs when the floating joints are being added to connect the newly added model to the tree. We can optimize this by keeping track of which links are part of the newly added model and only searching over these links. This issue tracks the implementation of this optimization.

sherm1, 2016-04-11:
Is this a performance problem? If it only occurs during modeling maybe the dumb method is good enough?

david-german-tri, 2016-04-11:
I think this is a reference to #2053.  It is indeed an issue at modeling time only, so the peformance problem is incidental.  The real issue is code clarity, and I'm not (yet) convinced it's OK to defer a fix, but we should probably keep discussion on the PR.

liangfok, 2016-04-11:
Yes, this is in reference to #2053. I will publish my reasoning to kick-the-can-down-the-road soon :).

PR #2053 (2016-04-19): Allow Arbitrary Initial Pose of Model in World Frame
Resolves https://github.com/RobotLocomotion/drake/issues/2003
Resolves #2058

# Key Changes

* Modified the semantics of `RigidBodyTree:addRobotFromFile()` to allow arbitrary positioning of a model's world frame within Drake's world frame.
* Added method `RigidBodyTree::AddFloatingJoint()` and modified `RigidBodyTreeURDF` and `RigidBodySDF` to use it. This consolidates what would otherwise be duplicate code between the URDF and SDF parsers.
* Added method `RigidBodyTree::AddRigidBody()` and modified `RigidBodyTreeURDF` and `RigidBodySDF` to  [... cut]
closed 2016-04-19 by RussTedrake