RobotLocomotion/drake #2074: Excessive system load compiling rigid body tree

mwoehlke-kitware, 2016-04-12:
Some of drake's source files take an excessive amount of time and memory to compile. `RigidBodyTree{,SDF,URDF}.cpp` are three particular offenders, requiring:

    RigidBodyTree.cpp     104 sec, 4.1 GiB
    RigidBodyTreeSDF.cpp   83 sec, 3.6 GiB
    RigidBodyTreeURDF.cpp  77 sec, 3.5 GiB

Since these files are closely related, and therefore have a tendency to get built at the same time by parallel builds, on systems with only 16 GiB of RAM (assuming some running background applications, such as an IDE and web browser), it's entirely possible for this trio to consume all available memory. In particular, my (current) main machine becomes effectively unusable for tens of minutes when this trio of files hits the compile queue due to thrashing.

The first source file can be split into pieces, which helps, but the latter two are much less amenable to this process. From some experimentation, the problem *seems* to be due to the various joint classes (commenting out all code except `#include "joints/DrakeJoints.h"` still takes 40 sec, 2.1 GiB).

The above numbers were produced using `/usr/bin/time --format=%U,%M` to measure hand-compiling the aforementioned files with `-g -O2` (roughly equivalent to `CMAKE_BUILD_TYPE=RelWithDebInfo`). Even with no debug/optimization flags, time and memory use is about half the above numbers, which is still fairly high.

david-german-tri, 2016-04-13:
Thanks!  This analysis is really helpful.  It's a pain point for us too, but as far as I know no one is looking at it yet, so I'll provisionally assign myself.

@liangfok, you may also be interested.

liangfok, 2016-04-13:
What does "joint classes" man in this context?

+1 for a redesign of the URDF / SDF parsers. I believe there should be a `RigiBodyTree` class that contains no parsers, and separate `DrakeURDFParser` and `DrakeSDFParser` classes that inherent from an abstract `ModelParser` class.

mwoehlke-kitware, 2016-04-13:
Thanks, @david-german-tri; it's good to know it's not just me. One of the reasons I wanted to open an issue is because I'm not sure what our minimum system requirements are for building drake, and I could imagine some "average" user with an underpowered machine getting bitten by this.

> What does "joint classes" mean in this context?

Subclasses of `DrakeJoint`. In particular, see those classes defined via includes in `drake/systems/plants/joints/DrakeJoints.h`.

I have a (very stale) branch (`REBASE-rbt-split-templates` in my fork) that splits `RigidBodyTree.cpp` into several pieces, with the worst needing only about 20 sec, 1.5 GiB to compile (most are about 10 sec, 1.0 GiB). Despite that this takes non-trivially more CPU cycles in total to compile, it's nearly a wash on higher-end machines, and does actually help on mid-range machines. (As an added bonus, the template instantiations are also much more legible.) However, it is not possible to do anything similar for `RigidBodyTree{SDF,URDF}.cpp`. The problem there seems to be with instantiating the various joint classes, which is why I'm not aware that anything can be done except to redesign those somehow so as to avoid the problem. (This is why I opened an issue rather than a PR; I don't have such a redesign available and wouldn't presume to be the best person to attempt such a change.) I expect that would also help `RigidBodyTree.cpp` considerably.

That said, I'd be happy to look at redoing the changes to split `RigidBodyTree.cpp` if you feel that would be valuable.

sherm1, 2016-04-13:
On Windows with VS 2015 RelWithDebInfo I'm seeing about 1.5GB in use when compiling those files. That's much bigger than a typical compile but still reasonable.

mwoehlke-kitware, 2016-04-25:
On a related note... I tried to do a build last Friday, and noticed 8 compile tasks at about 1.5 GiB each. Unfortunately, this broke my computer :cry: (read: caused it to become entirely non-responsive, and it did not recover after being left along over the weekend), so I can't do a postmortem to determine what it was trying to build at the time.

jwnimmer-tri, 2016-04-27:
The compiler grinding to a halt on these files is really starting to annoy me, too.  If there's a patch nearby to break out some code into separate files, that'd be a good start.  Once that's in, we could try to recruit someone at TRI to tackle the compiler-bogging problem with `drake/systems/plants/joints` code.

jwnimmer-tri, 2016-05-14:
Perhaps we should also see if this compiler flag can be removed, once this issue is fixed:
>  set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} /bigobj") # after receiving RigidBodyTree.cpp : fatal error C1128: number of sections exceeded object file format limit: compile with /bigobj

david-german-tri, 2016-06-16:
This has generated a bunch of complaints recently, and is a barrier to switching to Ninja, so I'm bumping it to `priority: high` and will start working on it instead of System2 for the next couple days.

david-german-tri, 2016-06-20:
The @mwoehlke-kitware WIP to split RigidBodyTree into pieces is here: https://github.com/mwoehlke-kitware/drake/commit/e7717a062366b81d3b400dddc863c77ce56151f4

jwnimmer-tri, 2016-06-28:
@david-german-tri Thanks for this!  Especially #2614 was a huge difference.  Even though more improvements are possible (and should continue), I wonder if we can call this ticket closed, or at least lower its priority?  I haven't needed more than even 8 GB during `-j8` builds since these changes.

david-german-tri, 2016-06-28:
There are two related that I've been working under the aegis of this ticket:

1. The three `RigidBodyTree` compilation units named in the OP are huge.

2. The `RigidBody*` headers pull a large amount of template code into hundreds of downstream compilation units that include them.

I agree with you that good progress has been made on item (2), but for item (1) we're still in the earlier stages.  I think breaking the dependencies on `drakeGeometryUtil` will make a huge difference.  So, I'd like to keep this ticket open until that's done, but I've dropped the priority to medium.

liangfok, 2016-06-28:
What about the plan to extract the URDF / SDF parsing code from `RigidBodyTree`? Is that going to part of this issue or should that be a separate issue?

david-german-tri, 2016-06-28:
I'm not planning to factor out URDF/SDF parsing as part of this issue.  It's a great idea for many reasons, but I'm not sure it will make a big difference to memory footprint.

jwnimmer-tri, 2016-06-29:
So, I retract my proposal to close this, and my high watermwark evidence of 8G upthread.  It turns out that when I switched to Ninja, I stopped compiling in Release mode (CMAKE_BUILD_TYPE was undefined).  Having switched back to release mode now, memory is topping out above 16GB even under `-j4` again.

mwoehlke-kitware, 2016-07-06:
We've gone backwards here somewhere...

    joints/RollPitchYawFloatingJoint.cpp  68 sec, 3.4 GiB
    parser_urdf.cc                       109 sec, 3.8 GiB
    RigidBodyTreeSDF.cpp                 108 sec, 4.5 GiB
    RigidBodyTree.cpp                    268 sec, 7.3 GiB

(This is from stats collected from the *entire* build¹, on Linux using GCC 4.9.2 in `RelWithDebInfo` mode. The good news is that the next worst offender is `examples/Quadrotor/runLQR.cpp` at a little under 2 GiB and 41 seconds, after which nothing exceeds 1.5 GiB or 31 seconds.)

(¹ Of drake itself. Stats for the externals were not collected.)

mwoehlke-kitware, 2016-07-06:
For grins, here's a plot (log scale) of build times to memory usage. Most of the build is in a reasonable region between 1-15 sec and about 150 KiB - 1 GiB.
[image]

liangfok, 2016-09-01:
# Benchmark Results

## System
* MacBookPro11,5 (mid 2015)
* Intel(R) Core(TM) i7-4980HQ CPU @ 2.80GHz
* 16 GB 1600 MHz DDR3
* Mac OS X, ProductVersion: 10.11.6, BuildVersion: 15G31

## Commands

```
$ cd drake-distro
$ rm -rf build
$ rm -rf externals
$ git reset --hard HEAD
$ cmake .. -DCMAKE_BUILD_TYPE:STRING=RelWithDebInfo -DDISABLE_MATLAB=TRUE -DWITH_DREAL=FALSE -DWITH_SNOPT=FALSE
$ time make -j4
```

Notice that I disable both MATLAB and dReal.

## SHA

[0442d21362dbd23b326238f8d190080f0aae248f](https://github.com/RobotLocomotion/drake/commit/0442d21362dbd23b326238f8d190080f0aae248f)

## Results

```
real   	11m17.079s
user   	34m42.356s
sys    	2m25.955s
```

david-german-tri, 2016-09-01:
@mwoehlke-kitware: Interesting.  I think that shows we've gone backwards in two respects:

1. It remains true that `DrakeJoints.h` (and in particular  `RollPitchYawFloatingJoint.h`) pull in a huge amount of template code.  We've factored compilation units that include `DrakeJoints.h` into multiple pieces.  However, as @liangfok's metrics point out, and indeed as you suggested upthread, it's not clear the step backwards has much effect on user-perceived build times for multicore systems with 16GB or more of RAM.  RigidBody* is a dependency chokepoint for the entire build, so when we hit these compilation-mega-units, there are plenty of cores to spare.

2. `RigidBodyTree.cpp` somehow picked up another 3+ GB of RAM usage.  That's insane and alarming; it should be bisected for root cause.

Right now, this issue is assigned to me, but I don't have bandwidth to work on it.  Someone else is welcome to chip in.

liangfok, 2016-09-01:
I'll take on the investigation of why `RigidBodyTree` takes so much memory to compile since it is very likely my fault. My current suspicion is that it happened when I extracted the parser code into their own `.h` and `.cc` files.

liangfok, 2016-09-03:
# Benchmarks 2

## Platform

* Lenovo T430
* Intel(R) Core(TM) i7-3520M CPU @ 2.90GHz
* 16GB DDR3 1600 MHz RAM

## Procedure
Here I wanted to try to replicate what @mwoehlke-kitware reported in the original description of this issue. To do this, I first build Drake using `VERBOSE=true` to get the actual compile commands. I then found the command for building `RigidBodyTree.cpp` and executed the following

```
$ cd /home/liang/dev/drake-distro-2/build/drake/systems/plants
$ rm CMakeFiles/drakeRBM.dir/RigidBody.cpp.o
$ $ /usr/bin/time --format=%U,%M /usr/bin/g++-4.9   -DHAVE_SPDLOG -DdrakeRBM_EXPORTS -I/home/liang/dev/drake-distro-2/drake/.. -I/home/liang/dev/drake-distro-2/build/drake/exports -I/home/liang/dev/drake-distro-2/build/install/include -I/home/liang/dev/drake-distro-2/build/drake -I/home/liang/dev/drake-distro-2/build/drake/lcmtypes -isystem /home/liang/dev/drake-distro-2/build/install/include/eigen3 -I/home/liang/dev/drake-distro-2/drake/thirdParty/bsd/spruce/include  -Werror=all -Werror=ignored-qualifiers -DGTEST_DONT_DEFINE_FAIL=1 -DGTEST_DONT_DEFINE_SUCCEED=1 -DGTEST_DONT_DEFINE_TEST=1 -O2 -g -DNDEBUG -fPIC -fvisibility=hidden -fvisibility-inlines-hidden   -std=gnu++14 -o CMakeFiles/drakeRBM.dir/RigidBodyTree.cpp.o -c /home/liang/dev/drake-distro-2/drake/systems/plants/RigidBodyTree.cpp
```

## Results
Here's what the last command above reported:
```
618.33,7894972
```
The above numbers indicate it took 618.33 seconds (~10 minutes) and 7,894,972 kilobytes of memory. The build type was `CMAKE_BUILD_TYPE:STRING=RelWithDebInfo`.

liangfok, 2016-09-05:
I extracted `RigidBodyTree::AddFloatingJoint()` and the joint times into their own `.h` and `.cc` files. See: https://github.com/liangfok/drake/tree/feature/extract_joint_types_and_add_floating_joints.

The overall benchmark results of https://github.com/liangfok/drake/commit/51e481dea44aecb13d2ba4547911a6a4b73fe53e show building Drake takes 10,258 user mode CPU seconds (2.85 minutes) and 5.1 GB on a Lenovo T430 laptop:

```
$ cd drake-distro
$ rm -rf build
$ rm -rf externals
$ git reset --hard HEAD
$ mkdir build
$ cd build
$ cmake .. -DCMAKE_BUILD_TYPE:STRING=RelWithDebInfo -DDISABLE_MATLAB=TRUE -DWITH_DREAL=FALSE -DWITH_SNOPT=FALSE 
$ /usr/bin/time --format=%U,%M   make -j4
10258.16,5124704
```

Using the current head of master ():

```
```

david-german-tri, 2016-09-06:
Great analysis and results!  I'd like to be a reviewer of this PR.  As a preview, here are my top three comments/questions from skimming your branch.

* Are we really sure that `AddFloatingJoint` belongs to `drake::parsers`?  That approach (a) is semantically weird if we ever need to add a floating joint outside a parser and (b) creates a bunch of new dependencies on RigidBodyTree public data.  Another option, which I also don't love, would be to leave AddFloatingJoint as a member of RigidBodyTree, and just move the implementation to a separate .cc file.  Maybe there is a third option?

* Factoring out the joint types into a separate header is a great idea.  It would help readability to also format and namespace them properly in this PR, e.g. `drake::systems::joints::kFixed`.

liangfok, 2016-09-06:
# Benchmarks 3

I use my HP z460 workstation to compare the latest head of master with the extraction of the floating base types and `RigidBodyTree::AddFloatingJoints()` into their own compilation units.

## Platform
* Intel(R) Xeon(R) CPU E5-2620 v3 @ 2.40GHz
* 

## Commands
Since the workstation as 12 cores, I use `-j12`.
```
$ cd drake-distro
$ rm -rf build
$ rm -rf externals
$ git reset --hard HEAD
$ mkdir build
$ cd build
$ cmake .. -DCMAKE_BUILD_TYPE:STRING=RelWithDebInfo -DDISABLE_MATLAB=TRUE -DWITH_DREAL=FALSE -DWITH_SNOPT=FALSE 
$ /usr/bin/time --format=%E,%U,%M make -j12
$ du -ch | grep total
```

## Results

Using the latest on master (8e6c585c73e573133a3cf76b02972e5c00ab433d):

```
$ /usr/bin/time --format=%E,%U,%M make -j12
16:29.95,4813.20,7903908
$ du -ch | grep total
6.1G	total
```

Using the optimized branch (https://github.com/liangfok/drake/commit/1bff65ce81aa9fe608eb23a87df46380e392a5e3):

```
$ /usr/bin/time --format=%E,%U,%M make -j12
13:13.46,4788.78,5097876
$ du -ch | grep total
6.1G	total
```

## Conclusions

Extracting `DrakeJoint::FloatingBaseType()` and `RigidBodyTree::AddFloatingJoint()` into their own compilation units decreased build times from ~16 minutes to ~13 minutes and, more importantly, decreased the maximum memory footprint to be from 7.9 GB to 5.1 GB.

liangfok, 2016-09-06:
> Are we really sure that AddFloatingJoint belongs to drake::parsers? That approach (a) is semantically weird if we ever need to add a floating joint outside a parser and (b) creates a bunch of new [...]

Yes, I believe `RigidBodyTree::AddFloatingJoint()` should be part of parsers since we only anticipate it ever being called by the parsers. Recall that it was originally added as a way to connect newly added models to an existing `RigidBodyTree`. It does this by searching through _all_ bodies in the tree for those that are parent-less, and adding floating joints to them. This can be thought of as a hack that was needed based on the limitations of the parsers (At the time, parsers couldn't keep track of which models they were adding. Now, with the introduction of the `model_instance_id`, this is no longer the case.). Longer-term, I expect the parsers to be able to automatically add the floating joints as it is parsing the model and adding a model instance, meaning this method will no longer be necessary.

Note that even if we remove this method from `RigidBodyTree`, we can continue to programmatically add floating joints using a combination of the following two methods:

* [RigidBodyTree::add_rigid_body()](http://drake.mit.edu/doxygen_cxx/class_rigid_body_tree.html#a9185736c83405cd991c2380500d813fe)
* [RigidBody::setJoint()](http://drake.mit.edu/doxygen_cxx/class_rigid_body.html#a418b727878b8a58d8b8e97282cf2d5cc)

Regarding the concern about introducing new dependencies, my longer term plan is to pull the parsers into their own library, which is then linked against the existing `drakeRBM` library. In other words, users will interact directly with the parsers rather than go through the `RigidBodySystem` and `RigidBodyTree` to add models to them.

> Factoring out the joint types into a separate header is a great idea. It would help readability to also format and namespace them properly in this PR, e.g. drake::systems::joints::kFixed.

Yeah, I agree. In the spirit of incremental PRs, however, I will probably not initially namespace the floating base types. I do have another WIP branch that adds name spaces. I'll submit that PR after the initial PR that brings down the memory footprint.

sherm1, 2016-09-07:
BTW, for another approach to adding floating joints see Simbody's [`MultibodyGraphMaker`](https://simtk.org/api_docs/simbody/latest/classSimTK_1_1MultibodyGraphMaker.html#details) class which is independent of the parser and independent of the multibody tree. It is structured as a utility that absorbs body and joint information (as obtained by a parser typically) and then spits out a spanning tree plus loop constraints design for building the multibody tree, including any needed floating joints. We could consider that more flexible approach at some point, although I agree that Liang's proposal is an improvement.

liangfok, 2016-10-09:
# Benchmarks 1

## System
* MacBookPro11,5 (mid 2015)
* Intel(R) Core(TM) i7-4980HQ CPU @ 2.80GHz
* 16 GB 1600 MHz DDR
* Ubuntu 14.04.1, clang-3.7.1, ninja

## Commands

```
$ cmake .. -G Ninja -DCMAKE_BUILD_TYPE:STRING=RelWithDebInfo -DDISABLE_MATLAB=TRUE
$  /usr/bin/time --format=%E,%U,%M ninja
```

### SHA

3af99f19790b2341cc4901fa871cacd6d14c7634

### Results

```
18:34.81,5875.01,3223556
```

It took about 18.5 wall clock minutes, 1.63 CPU hours, and 3.2GB of RAM.

jwnimmer-tri, 2017-01-26:
So here's a good way to profile the build:
```sh
$ bazel build --profile profile.bin //...
$ bazel analyze-profile profile.bin --dump=raw > profile.csv
$ grep 'ACTION_EXECUTE.*Compiling' profile.csv |
  sort -t\| -k5,5rn | head -n40 | cut -d\| -f5,8 | 
  perl -pe 's/^(\d+)\d{9}\|/\1s /;' | 
  awk '{if ((NR-1) % 2 ==0) print}'
```

For me with default Bazel config (GCC, release), it blames:
```
177s Compiling drake/multibody/rigid_body_tree.cc
116s Compiling drake/multibody/parsers/sdf_parser.cc
107s Compiling drake/multibody/parsers/urdf_parser.cc
95s Compiling drake/multibody/parsers/parser_common.cc
85s Compiling drake/multibody/test/rigid_body_tree/rigid_body_collision_clique_test.cc
64s Compiling drake/multibody/collision/test/collision_filter_group_test.cc
53s Compiling drake/multibody/joints/roll_pitch_yaw_floating_joint.cc
49s Compiling drake/multibody/joints/roll_pitch_yaw_floating_joint.cc
45s Compiling drake/common/test/symbolic_mixing_scalar_types_test.cc
36s Compiling drake/multibody/constraint/rigid_body_constraint.cc
35s Compiling drake/systems/analysis/test/runge_kutta3_integrator_test.cc
34s Compiling drake/multibody/rigid_body_plant/test/compute_contact_result_test.cc
31s Compiling drake/solvers/test/optimization_examples.cc
30s Compiling drake/solvers/test/optimization_examples.cc
28s Compiling drake/multibody/joints/quaternion_floating_joint.cc
26s Compiling drake/systems/framework/test/diagram_test.cc
26s Compiling drake/systems/analysis/test/simulator_test.cc
24s Compiling drake/systems/framework/test/diagram_builder_test.cc
24s Compiling drake/systems/controllers/test/pid_controlled_system_test.cc
23s Compiling drake/multibody/joints/quaternion_floating_joint.cc
```

jwnimmer-tri, 2017-03-31:
FYI my WIP branch on fixing this is https://github.com/jwnimmer-tri/drake/tree/rbt-build-time.  I haven't had a chance to correctly reprofile and tune up the results, but it's a solution framework.

jwnimmer-tri, 2018-05-01:
Hopefully with #8442 and #8543 merged, this is now "good enough".