RobotLocomotion/drake #2077: SimpleCar shows library link failures on Windows.

rpoyner-tri, 2016-04-13:
For example, see https://drake-jenkins.csail.mit.edu/job/experimental/798/ . SimpleCar demo and unit test build are currently disabled on Windows as a consequence.

david-german-tri, 2016-04-13:
Perhaps your library is not including the visibility macros required to get Windows to put symbols in a DLL?  (See the generated file `drakeRBM_export.h` for an example.)

david-german-tri, 2016-04-13:
Untagging Kitware because this is likely not a CI issue per se.

rpoyner-tri, 2016-04-13:
@david-german-tri it is far from clear to me how symbol visibility causes this:

`
  LINK : fatal error LNK1181: cannot open input file '..\..\lib\Release\simple_car.lib' [C:\projects\workspace\experimental\e052ff41\drake\pod-build\examples\SimpleCar\simple_car_demo.vcxproj]
`

However, I will continue to investigate.

david-german-tri, 2016-04-13:
Huh.  I wonder if a DLL is even created for a shared library that exports no symbols.  Contrast:

https://github.com/RobotLocomotion/drake/blob/master/drake/examples/Atlas/atlasUtil.h

If making the library static fixes it, that seems OK on general principles, but also uncomfortable because it's not done anywhere else in drake?

RussTedrake, 2016-04-13:
David is correct.  You must list the exports to generate the .lib file.  That will resolve it.

rpoyner-tri, 2016-04-13:
FTR, a DLL is apparently built:

`
  Link:

    C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\bin\link.exe /ERRORREPORT:QUEUE /OUT:"C:\projects\workspace\experimental\e052ff41\drake\pod-build\lib\Release\simple_car.dll" /INCREMENTAL:NO /NOLOGO /LIBPATH:"C:/projects/workspace/experimental/e052ff41/drake/pod-build/lib" /LIBPATH:"C:/projects/workspace/experimental/e052ff41/drake/pod-build/lib/Release" /LIBPATH:C:/projects/workspace/experimental/e052ff41/build/lib /LIBPATH:C:/projects/workspace/experimental/e052ff41/build/lib/Release kernel32.lib user32.lib gdi32.lib winspool.lib shell32.lib ole32.lib oleaut32.lib uuid.lib comdlg32.lib advapi32.lib /MANIFEST /MANIFESTUAC:"level='asInvoker' uiAccess='false'" /manifest:embed /PDB:"C:/projects/workspace/experimental/e052ff41/drake/pod-build/lib/Release/simple_car.pdb" /SUBSYSTEM:CONSOLE /TLBID:1 /DYNAMICBASE /NXCOMPAT /IMPLIB:"C:/projects/workspace/experimental/e052ff41/drake/pod-build/lib/Release/simple_car.lib" /MACHINE:X86 /SAFESEH  /machine:X86 /DLL simple_car.dir\Release\simple_car.obj

    simple_car.vcxproj -> C:\projects\workspace\experimental\e052ff41\drake\pod-build\lib\Release\simple_car.dll
`
but the later link step (already doomed for symbol visibility) looks for a .lib:

`LINK : fatal error LNK1181: cannot open input file '..\..\lib\Release\simple_car.lib' [C:\projects\workspace\experimental\e052ff41\drake\pod-build\examples\SimpleCar\simple_car_demo.vcxproj] [C:\projects\workspace\experimental\e052ff41\pod-build\drake.vcxproj]`

I'm not sure if that is a CMake bug, or if that is just the best result to be hoped for. In any case, PR #2084 addresses the present case, and https://blog.kitware.com/create-dlls-on-windows-without-declspec-using-new-cmake-export-all-feature/ offers hope for the future.

closed 2016-04-13 by rpoyner-tri

RussTedrake, 2016-04-13:
i believe that is the expected visual studio behavior.  the lib file contains the export symbol table for the dll.  if there are no symbols exported, the dll appears (not sure why) but the lib does not.