RobotLocomotion/drake #2078: Add build system support for long-running regression tests

jwnimmer-tri, 2016-04-13:
In #2068, Russ asked for:
> Is there a standard development solution for adding regression tests to the system
> which are not run for CI? For instance, it looks like we could restrict a test to only
> the nightly configuration in add_test . RobotLocomotion/kitware — what are the
> ctest recommended best practices for this?

(See a few comments atop there for initial discussion.)

The general idea here is that we want to write tests, but not have them run by default by developers, nor by PR builders.  We just want to run them on master CI, and let developers opt-in to running them if they want to explore something.

jwnimmer-tri, 2016-04-13:
From today's discussion, there is no way to teach CTest that a test should be off by default.  Kitware may file a CTest feature request to see if that can be made to happen.

Thus, so far these options to achieve this:
- Label the test and use CTest to opt out of specific labels.  This means that developers would need to have a longer CTest command line (or we have a drake wrapper script that encapsulates this option).
- Add a CMake option with an if-block around the add_test calls.  The downside is a trip through CMake to turn them on or off (maybe this is fine).

Feel to correct this if I got anything wrong.

jwnimmer-tri, 2016-05-17:
I believe that the LONG_RUNNING_TESTS has solved this?

Though perhaps _how_ to run the tests via CMake on your workstation still needs to be documented in developers.rst or something?

liangfok, 2016-05-17:
I believe the instructions on how to run unit tests are here and that they also apply to long running tests: http://drake.mit.edu/unit_testing_instructions.html

jwnimmer-tri, 2016-05-17:
Possibly, but did you try it?  I didn't think CTest knows the long running test names that match its `-R` argument unless you have manually configured LONG_RUNNING_TESTS=ON in your CMakeCache.

liangfok, 2016-05-17:
Oh yeah, you're right. Looks we need to run `cmake -DLONG_RUNNING_TESTS=ON ..` from within the build artifacts directory to get them to show up. I'm testing now and will issue a PR if this is correct.

sammy-tri, 2016-05-17:
I've only been able to figure out how to get the long running tests to show up by editing the CMakeCache (at @jwnimmer-tri 's suggestion).  A command line equivalent which could be invoked for a single ctest run would be interesting if it were doable.

liangfok, 2016-05-17:
I'm updating the wiki right now with instructions.  Please hold...

closed 2016-05-18 by david-german-tri