RobotLocomotion/drake #2080: Add USE_MATLAB build option, to force-disable matlab

jwnimmer-tri, 2016-04-13:
Per the discussion in #1860, starting at [this comment](https://github.com/RobotLocomotion/drake/issues/1860#issuecomment-204397872).

As with other optional prerequisites of Drake, we should have the ability to run a build and test while excluding matlab from the picture, even if it happens to be installed on the system.

jwnimmer-tri, 2016-04-13:
To @david-german-tri for triage.

david-german-tri, 2016-04-13:
I'm unconvinced this is solving a real problem, but if it makes the difference between people installing MATLAB and not I want to fix it regardless.

Some of Drake's external dependencies (notably LCM) run `find_program(matlab matlab)` in their own CMake files.  @jwnimmer-tri, will you be satisfied if Drake itself fails to find MATLAB (as with other optional dependencies), or do you require MATLAB to disappear from the entire transitive closure of the superbuild?

jwnimmer-tri, 2016-04-13:
Ideally, I'd like to be able to reproduce all of the #1923 supported Ubuntu configurations locally -- or perhaps more realistically, to at least within the measurement error of CI.

At the moment, drake/drake code has vastly different compile times, test outcomes, and possible failure modes in the with-matlab and without-matlab variants, which is my main interest in this ticket.

TTBOMK, the Ubuntu CI variants of with-matlab and without-matlab don't seem to differ in how LCM behaves (or really anything in the superbuild in general), so I'm fine with leaving LCM's matlab sensing alone.  Certainly I'm fine with just adding a matlab knob to drake/drake for now, and getting LCM in a second pass if it ever becomes a problem.

I'll also note that USE_MATLAB seemed to be the consensus on the other thread for the best way to make a build behave as if matlab wasn't installed.  If there's some other easier or better way (environment variables, schroot, treating unlicensed matlab as if matlab were missing, ...), I'm game for any recipe that is reliable.

jwnimmer-tri, 2016-04-13:
(The background here being that the main users of Drake that I will be supporting will not have matlab installed, so I want to test that configuration primarily -- but I also want to opt-in to matlab in order to ru the full CI suite locally when the need arises.)

sherm1, 2016-04-13:
> I'm unconvinced this is solving a real problem ...

For me the problem is that I use Matlab for various things myself so it is always on my machine. But a good part of my job is to get Drake to full functionality without Matlab -- it's going to be hard to claim success with that if I can't turn off Drake's use of Matlab!

david-german-tri, 2016-04-14:
OK.  It seems to me like the situations in which you'll need to reproduce a no-MATLAB result are rare, and you could accomplish it just by renaming your MATLAB executable temporarily.  (Haven't actually tested this yet, so perhaps `find_program` is much smarter than I'm assuming.)

Regardless, a USE_MATLAB flag for drake itself is pretty straightforward, so I'll go ahead and do that.  Thanks for the feedback!

sherm1, 2016-04-14:
> you could accomplish it just by renaming your MATLAB executable temporarily

I renamed the Matlab installation directory and that outsmarted find_program.
> a USE_MATLAB flag for drake itself is pretty straightforward

If there is a hard part, it is getting the message down to the externals in the superbuild.

billhoffman, 2016-04-14:
I just ran into this issue when trying to test the conversion to FindMatlab.cmake add_matlab_unit_test.  FindMatlab.cmake can find matlab without matlab being in the PATH.  This meant that in some cases matlab was found and some cases it was not.  This results is crashes and other odd things in the matlab tests. 

With a CMake superbuild project, the nice thing is that you can send flags and cache values from the superbuild into the children projects.  I would say we should come up with a way of finding/enabling matlab, then push that option equally down to the child projects via -D flags.   We could have a USE_MATLAB option in the toplevel superbuild CMakeLists.txt.  The default value of that option could be set to ON if matlab is in the PATH.   The next step would be passing the use of matlab and its location to all the projects build in the superbuild.

liangfok, 2016-04-14:
Not to change the subject, but I think the same mechanism enabling / disabling Matlab would be useful for enabling / disabling support for ROS in Drake.

liangfok, 2016-04-15:
Instead of adding a `USE_MATLAB` option at the super-build level, how about simply check for an environment variable like `DRAKE_INCLUDE_MATLAB_SUPPORT` that, when set to true, enables Matlab support?

billhoffman, 2016-04-15:
I don't like environment variables because they can change.  That is what the CMakeCache.txt file is for.  You set options for the build and they stay that way until you change them in the CMakeCache.txt file with ccmake or cmake-gui.  Environment variables can change, also cmake does not know to look at them each time the build is run, only when cmake is run.  So, CMake options are the place for things like this not env vars.

liangfok, 2016-04-15:
I see. OK thanks for the explanation.

PR #2109 (2016-04-19): Add a DISABLE_MATLAB option.
This option does not thwart other externals that want to look for MATLAB, but it does disable MATLAB in drake itself.

@jwnimmer-tri @sherm1 is this what you wanted?

Resolves #2080

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/2109) &emsp; Multiple assignees:&emsp;[image]&nbsp;<a href="/robotlocomotion/drake/pulls/assigned/david-german-tri">david-german-tri</a>,&emsp;[image]&nbsp;<a href="/robotlocomotion/drake/pulls/assigned/jonathan-owens">jonathan-owens</a>,&emsp;[image]&nbsp;<a href="/robotlocomotion/drake/pulls/assigned/lia [... cut]
closed 2016-04-19 by david-german-tri