RobotLocomotion/drake #2082: Require CMake 3.4 and update installation steps

david-german-tri, 2016-04-13:
CMake 3.4 would simplify and modernize the CMake lists.  Let's make it a requirement, and update the installation instructions appropriately.  (As discussed offline.)

rpoyner-tri, 2016-04-13:
Looks like CMake 3.4 would enable this, for example: https://blog.kitware.com/create-dlls-on-windows-without-declspec-using-new-cmake-export-all-feature/

.. and eventual removal of various declspec scars in source code and build systems.

Count me in.

jamiesnape, 2016-04-13:
We install 3.5 on all the CI machines for compatibility with CDash, so personally, I would say just require 3.5.

david-german-tri, 2016-04-13:
@jamiesnape Done.

wxmerkt, 2016-04-14:
I'm curious, what's the intended timeline/sequence of actions on this?

Ubuntu 16.04 appears to ship with 3.5.1, the most recent version on Ubuntu 14.04 is 2.8.12.2. Drake already requires adding a PPA for gcc-4.9 - this appears as requiring to add another non-standard PPA for CMake-3.x (maybe [this one here](http://askubuntu.com/questions/610291/how-to-install-cmake-3-2-on-ubuntu-14-04)). Or would it be possible to get CMake 3.5.x and gcc-4.9 from the same PPA?

jamiesnape, 2016-04-14:
No, and I would recommend downloading the binary from the CMake website. All dependencies are bundled so it works on any Linux distribution.

bradking, 2016-05-26:
With #2415 merged, this is now resolved.

closed 2016-05-26 by bradking