RobotLocomotion/drake #2085: accessing manipulatorDynamics in feedback/control

mjbeardsley, 2016-04-13:
I have a TimeSteppingRigidBodyManipulator robot with a PD feedback control loop, and I'd like to access the manipulatorDynamics function of the robot in the feedback loop. Rather than adding the huge matrices to the robot's output frame, is it possible for the PD DrakeSystems to access the manipulatorDynamics differently (e.g. passing the robot object by reference into the PD block DrakeSystem constructors)? 

I know that for classes, in order to pass-by-reference the class needs to subclass handle, but short of changing DynamicalSystem to subclass handle or adding the dynamics matrices to the robot's output frame, is there a better way for other blocks to access the robot manipulatorDynamics?

Thanks!

RussTedrake, 2016-04-13:
You can use getManipulator to get a copy for your PD controller.  It's true that it's pass by value, so acts like a copy, but if you don't modify it then the matlab backend seems to optimize this away pretty well.  You could probably even get away with calling getManipulator in the loop and not see a big performance hit.

mjbeardsley, 2016-04-13:
Ah yes this will work, I got a little mixed up about how getManipulatorDynamics worked. All good now though, thanks!

closed 2016-04-13 by mjbeardsley