RobotLocomotion/drake #2087: Compiler error on Ubuntu Linux with Nvidia Drivers: No rule to make target `/usr/lib/x86_64-linux-gnu/libGL.so'

liangfok, 2016-04-13:
I did a fresh install of Ubuntu Linux 14.04.4 64-bit desktop. Since my computer has an Nvidia GPU, I installed Nvidia's [proprietary Linux drivers (version 361.42)](http://www.nvidia.com/download/driverResults.aspx/101423/en-us).  I then followed Drake's [Ubuntu Linux installation instructions](http://drake002.csail.mit.edu/drake/sphinx/ubuntu.html) but ran into the following error while compiling:

```
make[10]: *** No rule to make target `/usr/lib/x86_64-linux-gnu/libGL.so', needed by `lib/libbot2-lcmgl-renderer.so.1'.  Stop.
```

The problem is `/usr/lib/x86_64-linux-gnu/libGL.so` points to `/usr/lib/x86_64-linux-gnu/mesa/libGL.so`, which points to `/usr/lib/x86_64-linux-gnu/mesa/libGL.so.1.2.0`, which does not exist.

Searching on my computer, I found a `/usr/lib/libGL.so.1`, which points to `/usr/lib/x86_64-linux-gnu/libGL.so.1`, which points to `/usr/lib/x86_64-linux-gnu/libGL.so.361.42`.

To fix the problem, I updated the symbolic link as follows:

```
$ sudo rm /usr/lib/x86_64-linux-gnu/libGL.so
$ sudo ln -s /usr/lib/libGL.so.1 /usr/lib/x86_64-linux-gnu/libGL.so
```

With the above fix, I was able to compile Drake. However, I'm not sure if the above fix is correct. If it is, we may need to add this procedure to Drake's installation instructions.

david-german-tri, 2016-05-24:
Losing `libGL.so` on Ubuntu seems to be a common phenomenon.  Google turns up issue threads similar to this one in a wide variety of open-source projects, with a variety of video drivers.  I don't think we can usefully capture any Drake-specific guidance.

closed 2016-05-24 by david-german-tri

soonho-tri, 2016-12-15:
FWIW, I had the same issue today. Thanks @liangfok for the solution!

raj-pandey, 2017-03-03:
I had the same problem. linker was taking /usr/lib/x86_64-linux-gnu/libGL.so was did had the definition for the "_glapi_tls_Dispatch". When I followed @liangfok suggestion and pointed to the correct file, compilation went ahead. Thumbs up for the solution.

stevenjj, 2017-07-05:
This worked!
I was facing similar issues with NVIDIA's proprietary driver and your two line command fixed the problem for a catkin build.
 
As a useful note, I was using the proprietary driver `NVIDIA-Linux-x86_64-381.09`  on Ubuntu 14.04

I had a lot of pre-installed packages and software(`srcsim`, `openpose_ros`, `cuda`, `cudann`, `ros-indigo-desktop` (without `ros-indigo-octomap`), and `pcl_ros') and was unsure whether one or more previously installed software caused linking problems.

Athanaze, 2019-02-10:
Thanks @liangfok ! This solution worked  for me, on Linux Mint.

rearming, 2019-12-18:
Thanks a lot!

mywnajsldkf, 2020-02-14:
Hi! I have a same issue like that. I tried your solution but I couldn't solve it..! 
Here's my error. 
```
make[2]: *** No rule to make target '/usr/lib/aarch64-linux-gnu/libGL.so', needed by 'lib/libopencv_viz.so.3.4.0'.  Stop.
CMakeFiles/Makefile2:3980: recipe for target 'modules/viz/CMakeFiles/opencv_viz.dir/all' failed
```

mywnajsldkf, 2020-02-14:
> Hi! I have a same issue like that. I tried your solution but I couldn't solve it..!
> Here's my error.
> 
> ```
> make[2]: *** No rule to make target '/usr/lib/aarch64-linux-gnu/libGL.so', needed by 'lib/libopencv_viz.so.3.4.0'.  Stop.
> CMakeFiles/Makefile2:3980: recipe for target 'modules/viz/CMakeFiles/opencv_viz.dir/all' failed
> ```
> 
> ubuntu18.04, ros-melodic

Oops ! i solved it I had a wrong path..

omerfy, 2020-03-30:
> Oops ! i solved it I had a wrong path..

How the path must be? I have the same problem and I changed the path as 

- /usr/lib/aarch64-linux-gnu/libGL.so 

- /usr/lib/libGL.so.1

but it didn't work.

duythaomta, 2020-05-22:
I ran the command: make -j2 and got this errror. Could you help me solve this errror. Thanks

make[2]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/libGL.so', needed by 'lib/libopencv_viz.so.3.4.4'.  Stop.
CMakeFiles/Makefile2:6204: recipe for target 'modules/viz/CMakeFiles/opencv_viz.dir/all' failed
make[1]: *** [modules/viz/CMakeFiles/opencv_viz.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 21%] Built target opencv_surface_matching
[ 24%] Built target opencv_imgproc
Makefile:162: recipe for target 'all' failed
make: *** [all] Error 2