RobotLocomotion/drake #2090: CMake Windows-specific syntax error during build

sherm1, 2016-04-13:
I'm trying @mwoehlke-kitware's much-improved CMake system from PR #1884 on Windows. I ran the CMake GUI, then built in VS 2015 using `build/drake-superbuild.sln`. I got this error:
>      Invalid escape sequence \U

This is due to a Windows-style "backslash" file name having been inserted into a CMake string in a generated file, without escaping the backslashes. Here is the CMake message:
```
CMake Warning (dev) at lcm/cmake_install.cmake:65 (file):
Syntax error in cmake code at
    C:/Users/Sherm/Documents/GitHub/sherm1/drake-distro/build/externals/lcm/lcm-prefix/src/lcm-build/lcm/cmake_install.cmake:65
when parsing string
    C:\Users\Sherm\Documents\GitHub\sherm1\drake-distro\build\externals\lcm\lcm-prefix\src\lcm-build\lib\pkgconfig/lcm.pc

     Invalid escape sequence \U

-- Installing: C:/Users/Sherm/Documents/GitHub/sherm1/drake-distro/build/lib/pkgconfig/lcm.pc
```
Here is an extract from the `lcm/cmake_install.cmake` file near line 65 where the error occurred (the full name is `C:/Users/Sherm/Documents/GitHub/sherm1/drake-distro/build/externals/lcm/lcm-prefix/src/lcm-build/lcm/cmake_install.cmake`):
```
### These are fine (forward slashes)
if(NOT CMAKE_INSTALL_COMPONENT OR "${CMAKE_INSTALL_COMPONENT}" STREQUAL "Unspecified")
  file(INSTALL DESTINATION "${CMAKE_INSTALL_PREFIX}/include/lcm" TYPE FILE FILES
    "C:/Users/Sherm/Documents/GitHub/sherm1/drake-distro/externals/lcm/lcm/lcm-src/lcm/eventlog.h"
    "C:/Users/Sherm/Documents/GitHub/sherm1/drake-distro/externals/lcm/lcm/lcm-src/lcm/lcm.h"
    )
endif()

### This one has backslashes (that's line 65)
if(NOT CMAKE_INSTALL_COMPONENT OR "${CMAKE_INSTALL_COMPONENT}" STREQUAL "Unspecified")
  file(INSTALL DESTINATION "${CMAKE_INSTALL_PREFIX}/lib/pkgconfig" TYPE FILE FILES
  "C:\Users\Sherm\Documents\GitHub\sherm1\drake-distro\build\externals\lcm\lcm-prefix\src\lcm-build\lib\pkgconfig/lcm.pc")
endif()

### This one is fine
if(NOT CMAKE_INSTALL_LOCAL_ONLY)
  include("C:/Users/Sherm/Documents/GitHub/sherm1/drake-distro/build/externals/lcm/lcm-prefix/src/lcm-build/lcm/windows/cmake_install.cmake")
endif()
```

@mwoehlke-kitware -- assigning to you first in case it is related to #1884.

mwoehlke-kitware, 2016-04-14:
Hmm... at first blush, it looks like the LCM pod is broken if using non-Cygwin CMake on a system that has Cygwin (specifically, `cygpath`) "visible" (e.g. in `%PATH%`; not sure if `pods.cmake` also has logic to find it even if not in `%PATH%`).

billhoffman, 2016-04-14:
Seems that this issue has come back in one form or another: https://github.com/RobotLocomotion/drake/issues/1676

mwoehlke-kitware, 2016-04-14:
Okay... more specifically, I think this is what is happening. In `pods_config_search_paths`, which initially sets `PKG_CONFIG_OUTPUT_PATH` and later calls `pkg_config_path(PKG_CONFIG_OUTPUT_PATH)`, converting the path to `\`s. Later, this gets injected into `pc_fname` which eventually is the source for an `install(FILES)`.

It looks like my initial assessment was backwards; there is a later call to `shell_path(PKG_CONFIG_OUTPUT_PATH)` that is *supposed* to fix this, but actually does nothing because `cygpath` is *not* present. I think I have a fix, but need to compare notes with @billhoffman how to get it tested and pushed, as it may require changes to three repositories to fully propagate.

PR #2202 (2016-04-28): Update 'cmake' external (also in 'lcm')
Update to newer version of `cmake` module that contains a `pkg-config` related fix for Windows. Update to a newer `lcm` that is likewise using the new version of `cmake` (which it duplicates as a git submodule), and also switch back to the "main" repository for the same.

Fixes #2090.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/2202)
commit 924373cb (2016-04-28): Update 'cmake' external (also in 'lcm')

Update to newer version of 'cmake' module that contains a pkg-config
related fix for Windows. Update to a newer 'lcm' that is likewise using
the new version of 'cmake' (which it duplicates as a git submodule), and
also switch back to the "main" repository for the same.

Fixes #2090.
closed 2016-04-28 by david-german-tri