RobotLocomotion/drake #2106: Upgrade to Eigen 3.3 asap

sherm1, 2016-04-15:
A bug in Eigen's `result_of` struct as described [here](http://eigen.tuxfamily.org/bz/show_bug.cgi?id=96) causes alignment problems in 32 bit Windows builds. This shows up in the Quadrotor example at least, which is currently excluded from 32 bit Windows builds.

david-german-tri, 2016-04-15:
I think you're in the best position to confirm this fixes your problem locally, so I tagged you as assignee.  Feel free to undo that if you disagree.  :-)

RussTedrake, 2016-04-24:
@sherm1 -- i suspect this one could be due to all the make_shared allocations flying around, too.  you're missing the aligned allocation.  do you have a reason to believe differently?

sherm1, 2016-04-24:
> i suspect this one could be due to all the make_shared allocations flying around, too. you're missing the aligned allocation. do you have a reason to believe differently?

Yes. At your earlier suggestion I tried aligned allocation but still couldn't build on 32 bit. Tracing the failure led to the `result_of` bug mentioned above. It is likely that both that fix and aligned allocation will be required, but I decided not to make any changes now since I can't test that they work on 32 bit. No change to Quadrotor was required to get it to work on 64 bit Windows.

liangfok, 2016-04-24:
@sherm1: What is the error that you see when running `runQuadrotorDynamics.exe`? I just tried compiling Drake against the latest development version of Eigen, which includes [this patch](https://bitbucket.org/eigen/eigen/commits/f2e6b911501d), and see the following error when trying to run this application:

[image]

I'm wondering whether this behavior is any different than when you last attempted to run it.

sherm1, 2016-04-24:
Liang, I *think* that may be due to Quadrotor's use of `make_shared`. Those should be replaced by `allocate_shared` -- see discussion above.

sherm1, 2016-04-24:
The error message I was getting is included in the discussion in now-defunct PR #1849. It starts like this:
```
eigen3\eigen\src/Core/CwiseBinaryOp.h(54): error C2719: 'unnamed-parameter': 
formal parameter with requested alignment of 16 won't be aligned
```

liangfok, 2016-04-24:
Great! Since by switching to the latest development version of Eigen resulted in me not seeing the `'unnamed-parameter': 
formal parameter with requested alignment of 16 won't be aligned` error, I have high confidence that the eventual upgrade to Eigen 3.3 will resolve this issue.

liangfok, 2016-04-24:
I tried replacing all `make_shared` instances with `allocated_shared` in this commit: https://github.com/liangfok/drake/commit/0441bfe77604bdeca001403f78c5c480a07a987b

Unfortunately, the memory alignment problem persists when running the quad rotor dynamics demo on a 32-bit version of Windows.

Note that this is built against the latest development version of Eigien that includes the `result_of` bug fix.

sherm1, 2016-04-25:
@liangfok, I've been trying to get my 32 bit Windows build working again but I keep running into this problem (25 times):
```
21>  "C:\Users\Sherm\Documents\GitHub\sherm1\drake-distro-build-d\drake\ALL_BUILD.vcxproj" (default target) (1) ->
21>  "C:\Users\Sherm\Documents\GitHub\sherm1\drake-distro-build-d\drake\systems\plants\test\testAccelerometer.vcxproj" (default target) (56) ->
21>  (Link target) ->
21>gtest.lib(gtest.dll) : fatal error LNK1112: module machine type 'x64' conflicts with target
machine type 'X86' 
[C:\Users\Sherm\Documents\GitHub\sherm1\drake-distro-build-d\drake\systems\plants\test\testAccelerometer.vcxproj]
```
I wiped out everything and built with the Win32 generator but it looks like it is using a Win64 generator for gtest. Have you seen that? And if so do you know a cure?

liangfok, 2016-04-25:
@sherm1: I haven't not encountered this error. Below is what I see on my machine. Maybe you can verify that your machine matches mine? This might be an issue where the PATH includes a directory to another `gtest.dll` somewhere on your computer.

Location of `gtest.dll` within my directory containing Drake build artifacts:
```
$ cd [directory containing drake build artifacts]
$ find . -name "gtest.dll"
./externals/googletest/googlemock/gtest/Debug/gtest.dll
./install/lib/gtest.dll
```

File types of found `gtest.dll` files:
```
$ file ./externals/googletest/googlemock/gtest/Debug/gtest.dll
./externals/googletest/googlemock/gtest/Debug/gtest.dll: PE32 executable (DLL) (console) Intel 80386, for MS Windows

$ file ./install/lib/gtest.dll
./install/lib/gtest.dll: PE32 executable (DLL) (console) Intel 80386, for MS Windows
```
Value of my `$PATH` environment variable:
```
$ echo $PATH
/c/Users/liang/bin:/mingw64/bin:/usr/local/bin:/usr/bin:/bin:/mingw64/bin:/usr/bin:/c/Users/liang/bin:/c/ProgramData/Oracle/Java/javapath:/c/Windows/system32:/c/Windows:/c/Windows/System32/Wbem:/c/Windows/System32/WindowsPowerShell/v1.0:/c/Users/liang/.dnx/bin:/c/Program Files/Microsoft DNX/Dnvm:/c/Program Files (x86)/Windows Kits/8.1/Windows Performance Toolkit:/c/Program Files/Microsoft SQL Server/130/Tools/Binn:/c/Program Files/Java/jdk1.8.0_77/bin:/c/Program Files/pkg-config-lite-0.28-1/bin:/c/Program Files (x86)/CMake/bin:/c/Program Files (x86)/GnuWin32/bin:/cmd:/c/Program Files (x86)/Windows Kits/10/Windows Performance Toolkit:/c/Users/liang/dev/drakebuild/drake/lib/Debug:/c/Users/liang/dev/drakebuild/install/lib:/usr/bin/vendor_perl:/usr/bin/core_perl
```
Note that the above looks like a bash shell even though it's on Windows because I'm using the bash terminal that comes with git, which is available [here ](https://git-scm.com/download).

sherm1, 2016-04-25:
Thanks, Liang. Did you build using the Visual Studio 2015 generator or one of the MinGW ones?

liangfok, 2016-04-25:
I used the Visual Studio 2015 generator:

[image]

I then compiled it from a terminal line using the command that I believe you taught me:

```
$ cd [build artifacts directory]
$ cmake --build . --target ALL_BUILD --config Debug
```

david-german-tri, 2016-06-23:
I think this may be cropping up again on #2593: https://drake-jenkins.csail.mit.edu/job/experimental/2197/compiler=msvc-ninja-32,label=windows/consoleFull.  Search for `C2718`.  Trying to confirm now.

tkoolen, 2016-08-26:
Related: https://github.com/RobotLocomotion/drake/issues/3211.

closed 2016-08-30 by tkoolen