RobotLocomotion/drake #2108: Rename CMake Option "WITH_[name]" to be Something More Accurate

liangfok, 2016-04-15:
At the super-build level, `[drake-distro]/CMakeLists.txt` contains numerous options named "WITH_[name]". If `ON`, Drake will install an external by itself. It is not meant to enable / disable features in Drake. Thus, we should come up with a better more descriptive name for these options. I suggest "INSTALL_[NAME]". Thoughts?

Related: https://github.com/gerkey/ros1_external_use/issues/12

liangfok, 2016-04-15:
Perhaps "INSTALL_[NAME]" is too frightening since someone may think the external software package will be installed in a system-wide fashion. Thus, how about: "INSTALL_IN_DRAKE_SUPERBUILD_[NAME]"?

RussTedrake, 2016-04-15:
see also #1523

jamiesnape, 2016-04-15:
`INSTALL_*` is not correct, anyway. The external will neither download nor build, and it does not really install in either case. `WITH_*` and `USE_*` are pretty common names for such CMake variables.

liangfok, 2016-04-17:
Aren't the externals being downloaded (and thus installed) into `[drake distro]/externals/[external name]/` when running `make` or `make download-all` at the super-build level?

jamiesnape, 2016-04-18:
Downloaded and installed are not equivalent in CMake.

liangfok, 2016-04-18:
Since the CMake world likes `USE_*` and the POD world actually downloads the externals, how about let's use `DOWNLOAD_AND_USE_*` as the variables in Drake's super-build level?

jamiesnape, 2016-04-18:
Maybe. Personally, I would table this until we have decided if/when/how we are going to replace the current external build system. If we use a pre-existing solution, the naming convention may be dictated by that.

liangfok, 2016-04-18:
OK I agree and will re-open this issue when / if the time is right.

closed 2016-04-18 by liangfok