RobotLocomotion/drake #2114: msan reporting uninitialized values in Bullet

david-german-tri, 2016-04-17:
This is likely another compile/link flag issue, but needs investigation.  See #2016 for history.

https://drake-cdash.csail.mit.edu/viewDynamicAnalysisFile.php?id=2718

jamiesnape, 2016-04-18:
Why do you think it is a linker issue?

david-german-tri, 2016-04-18:
No reason other than that it's in an external.  I hope to actually look at it today.

david-german-tri, 2016-04-18:
Aside: We currently have 18 msan failures like this one in `getPairCache`, 5 from [lcm_parse_url](https://drake-cdash.csail.mit.edu/viewDynamicAnalysisFile.php?id=2731), and 1 [assertion failure](https://drake-cdash.csail.mit.edu/viewDynamicAnalysisFile.php?id=2719)

sherm1, 2016-05-25:
@amcastro-tri, does this look like a bug in Bullet to you?

amcastro-tri, 2016-05-25:
I would need to investigate further. I'll talk to @david-german-tri since he was involved in this.

jwnimmer-tri, 2018-12-21:
The original report is no longer available from CDash, and we're not planning on using Bullet in new code.  No reason to keep this open.

closed 2018-12-21 by jwnimmer-tri