RobotLocomotion/drake #2115: Nested Class CollisionElement Not Supported

liangfok, 2016-04-18:
Here's the error being reported by the compiler:

```
drake/systems/plants/RigidBody.h:142: Warning 325: Nested class not currently supported (CollisionElement ignored)
```

I believe the best fix for this is to move CollisionElement into its own .h/cc files.

sherm1, 2016-04-18:
I agree this should be moved. @amcastro-tri is working to clean up this mess. But I'm curious as to whether this is a new problem and what compiler is complaining about nested classes -- those have been supported for a long time.

liangfok, 2016-04-18:
I don't believe it's a new problem because I recall seeing the warning for months now. Here's the compiler I'm using that resulted in the above warning being printed:

```
$ /usr/bin/gcc-4.9 --version
gcc-4.9 (Ubuntu 4.9.3-8ubuntu2~14.04) 4.9.3
```

liangfok, 2016-04-19:
**Update:** I also see this warning on OS X. 

```
$ clang --version
Apple LLVM version 7.3.0 (clang-703.0.29)
Target: x86_64-apple-darwin15.4.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```

jwnimmer-tri, 2016-04-25:
This is a swig warning, not a compiler warning.  I don't think it matters, other than possibly suppressing it to tidy up the default build output.

jwnimmer-tri, 2016-07-28:
I am going to take a crack at this; the CDash board is always polluted by this warning.

PR #2963 (2016-07-28): Extract RigidBody::CollisionElement into its own file
This extracts `RigidBody::CollisionElement` into its own files `rigid_body_collision_element.{h,cc}` and class `RigidBodyCollisionElement`, and updates all of the call sites to match.  Since the nesting wasn't supported by SWIG anyway, I don't expect this to change the python or matlab APIs at all.

Fixes #2115.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/2963)
closed 2016-07-28 by jwnimmer-tri