RobotLocomotion/drake #2121: System Input/Output Vector TypeDef "LCMMessageType" Needs New Name

liangfok, 2016-04-19:
Currently, the `InputVector`, `OutputVector`, and `StateVector` of a system in Drake must contain a `typedef` called `LCMMessageType`. This `typedef` defines the type of the data received, sent, or used by the system. It unfortunately couples the system to a particular transport layer. For example, classes [PendulumState](https://github.com/RobotLocomotion/drake/blob/b6257b940f92a972a6f9305e951cfb8fa659598f/drake/examples/Pendulum/Pendulum.h#L11) and [PendulumInput](https://github.com/RobotLocomotion/drake/blob/b6257b940f92a972a6f9305e951cfb8fa659598f/drake/examples/Pendulum/Pendulum.h#L52) are both specific to the LCM transport layer since they contain the following definition:

```
typedef drake::lcmt_drake_signal LCMMessageType;
```

This issue tracks the process by which systems in Drake are decoupled from the underlying transport layer.

david-german-tri, 2016-04-19:
This is gone in the System2 design.  I'd suggest it's not worth fixing in System1.

https://docs.google.com/a/tri.global/document/d/1YQIh2LUETE4CrQ6NvVeNefiIgXch5Q9JO0OWPyMOe90/pub

liangfok, 2016-04-19:
Great! That's what I suspected. Will close this issue now.

closed 2016-04-19 by liangfok