RobotLocomotion/drake #2127: Odd error message from snoptmex

samkhal, 2016-04-20:
I'm running into an odd error message from snoptmex: "F has incorrect row dimensions 101. Should be length 51". However, none of the variables that are getting passed to it have those dimensions, as you can see from my workspace. I'm not sure this has to do with Drake, but was wondering if anyone knew what might be causing this.

This doesn't relate to Atlas; I'm trying to run planFootsteps for my own biped.

 [image]

jwnimmer-tri, 2016-04-21:
There's some chance that the now-merged #2133 fix is related?

RussTedrake, 2016-04-21:
Actually, those changes are completely independent for now.

hongkai-dai, 2016-04-21:
It is complaining that the argument of your `nonlcon` has incorrect dimension. It probably has dimension 101 or 100. Could you evaluate your `nonlcon` at `x0` and check the return arguments?

samkhal, 2016-04-22:
I've since adjusted unrelated parameters that led to this error disappearing, and I can't seem to reproduce it again. I'll close the issue for now, then.

closed 2016-04-22 by samkhal