RobotLocomotion/drake #2129: Unable to reproduce linux-gcc-continuous-debug CI failure

sammy-tri, 2016-04-20:
Starting with build #53 https://drake-jenkins.csail.mit.edu/job/linux-gcc-continuous-debug/53/ testOptimizationProblem has been failing in the linux-gcc-continuous-debug builder.

I've been attempting to reproduce this failure locally by building Drake with BUILD_TYPE=Debug along with WITH_SNOPT and all other options set to their defaults.  Build system is Ubuntu 14.04 using g++ 4.9.  In this configuration, testOptimizationProblem passes reliably.

I'm not sure if the CI is configured strangely or if there's a lack of documentation/automation for Drake developers to correctly run the same tests as CI, but it should be possible to reproduce and debug CI failures locally.

david-german-tri, 2016-04-20:
Shot in the dark: could you try enabling all the externals?  Kitware switched at least the Linux CI to do that at some point.  (I'll update the documentation.)

sammy-tri, 2016-04-20:
This turns out to be #1935 all over again, basically.  I was not adequately turning on Debug.  Namely, just because "make options" tells you that you're configuring for a debug build, later invocations of "make" still call cmake with --config "Release" unless you make sure to always have BUILD_TYPE=Debug on every invocation of make.

jwnimmer-tri, 2016-04-21:
This issue now turns into: please, someone document in developers.rst (or related) how to set up and invoke a Debug-mode build on Ubuntu.  If it has to wait for the pod-build #1884-like fixes to complete, or the CI scripts to hit the drake-ci repo, or whatever, so be it.

liangfok, 2016-04-21:
Do the debug mode instructions on this page not work?http://drake002.csail.mit.edu/drake/sphinx/from_source.html

jwnimmer-tri, 2016-04-21:
One confusing part is that `make options` offers the user the option to change from `Release` to `Debug`.  Users can change that knob, but it doesn't actually work.  (This is what bit @sammy-tri.)

The `BUILD_TYPE=Debug` environment variable in those docs _does_ seem to work (I'm still testing some variations on it now).  So maybe that is enough.

However, the second confusing part is to that _you need to set it in your environment for every single make command you ever run_, because the build system does not notice when you change it, so it leaves the old release-mode object files intact and in effect.

At this point, the list of environment variables you have to set on Ubuntu in the default installation is a bit insane.  We can document it, sure, but it would be better to have the build system do the correct thing out of the box.

Even better would be if the build system _prevented_ you from firing a shotgun at your foot, so that users continue to have the use of both feet, even if they forgot to consult the build system documentation on that particular day.

liangfok, 2016-04-21:
+1 for making any knobs we provide through `ccmake` and any other mechanism intuitively work.

jamiesnape, 2016-04-21:
This is another symptom of the non-standard build system, unfortunately. With a normal CMake build something like `cmake -DCMAKE_BUILD_TYPE=Debug .` or just selecting `Debug` once in `ccmake` or `cmake-gui` is all you would need.

jwnimmer-tri, 2016-04-21:
Is the problem that we need to completely finish the work like #1884, or just that we need to tell users to run `cmake` instead of `make`?  If there are instructions fixes that can help in the near term, that's a great start.  I guess at a minimum, we could document setting that environment variable forever, but I'm hoping there's a more conventional answer that is more user-friendly and that works now.

jamiesnape, 2016-04-21:
The former (and once that is done you would call `cmake`/`ccmake`).

david-german-tri, 2016-05-25:
OK, so, everyone agrees this is documented correctly for the Makefile build.  Post-#1884, it also works for me in the quasi-out-of-source CMake build with `-DCMAKE_BUILD_TYPE=Debug`, but we aren't making that the default/recommended way to build Drake until it's fully out of source.  (In the current state, things get very broken if you try to switch between the quasi-out-of-source build and the Makefile build.)

I think there is nothing actionable left in this issue, so I'm closing it.  Feel free to reopen if you disagree.

closed 2016-05-25 by david-german-tri