RobotLocomotion/drake #2130: Some Unit Tests Employ Random Irreproducible Test Cases

liangfok, 2016-04-20:
Two unit tests that come to mind are:

1. drake/systems/plants/test/compareRigidBodySystems.cpp
2. drake/util/test/testDrakeGeometryUtil.cpp

There may be more. All unit tests should be deterministic.

@jwnimmer-tri I'll assign this to you so you can dole it out to the appropriate person to fix (feel free to bounce it back to me).

jwnimmer-tri, 2016-04-20:
While I appreciate the sentiment, I am disinclined to do anything about this until (1) the test is otherwise being functionally enhanced, which is a good time to fix additional deficiencies, or (2) if it is flaking a lot in CI.

liangfok, 2016-04-20:
OK I shall close this until the time is right.

closed 2016-04-20 by liangfok