RobotLocomotion/drake #2131: some matlab tests are failing after switching to cmake add_test

billhoffman, 2016-04-20:
They seem to all have this in common:
Undefined variable "drake" or class "drake.util.BotLCMGLClient".  Any idea what causes that? @RussTedrake 

Here are the failed tests in the continous builds:
https://drake-cdash.csail.mit.edu/index.php?project=Drake&showfilters=1&filtercount=2&showfilters=1&filtercombine=and&field1=label&compare1=61&value1=jenkins-linux-gcc-continuous-matlab-86&field2=buildstarttime&compare2=84&value2=now

Here there were passing in the CI build for the branch:
https://drake-cdash.csail.mit.edu/viewTest.php?onlypassed&buildid=10496

RussTedrake, 2016-04-20:
Looks like the java classpath is not getting set properly.

billhoffman, 2016-04-20:
And as if by magic, it started working....   https://drake-cdash.csail.mit.edu/index.php?project=Drake&showfilters=1&filtercount=2&showfilters=1&filtercombine=and&field1=label&compare1=61&value1=jenkins-linux-gcc-continuous-matlab-87&field2=buildstarttime&compare2=84&value2=now

billhoffman, 2016-04-20:
Where/how does the java class path get set for matlab tests?

RussTedrake, 2016-04-20:
in addpath_drake.m, starting around here:
https://github.com/RobotLocomotion/drake/blob/master/drake/addpath_drake.m#L100

david-german-tri, 2016-06-08:
This appears to have no remaining actions, so I'm going to close this and let the mystery lie.

closed 2016-06-08 by david-german-tri