RobotLocomotion/drake #2135: Error creating or communicating with child process

jwnimmer-tri, 2016-04-20:
The windows continuous builders are spewing "Error creating or communicating with child process" left and right, for example:
- https://drake-jenkins.csail.mit.edu/view/Continuous/job/windows-msvc-32-continuous-release/139/
- https://drake-jenkins.csail.mit.edu/view/Continuous/job/windows-msvc-32-continuous-release/140/

david-german-tri, 2016-04-21:
This is also happening to all experimental builds, and therefore is blocking all Drake development.  

Visual studio error code is `D8040`.  Documentation is pretty well nonexistent.  Compiler bugs :tada:

david-german-tri, 2016-04-21:
@jamiesnape I'm trying to reproduce this.  I've got a local Windows laptop, but it would be better if I could launch and remote into a buildslave VM.  Do you happen to have a recipe for doing that?

Also, if you have any thoughts on the issue itself they would of course be appreciated.  :-)

jamiesnape, 2016-04-21:
Log into https://us-west-1.console.aws.amazon.com/ec2/v2/home?region=us-west-1 and launch an instance from ami-1ce29d7c. It needs to be in us-west-1 because of issues with the Jenkins EC2 plugin.

jamiesnape, 2016-04-21:
@billhoffman thinks it may be due to a Windows compiler flag.

billhoffman, 2016-04-21:
OK, this is most likely the -MP flag I added last night to speed up the builds.  I will remove it.

billhoffman, 2016-04-21:
Just pushed a new build script up.   Lets see if the next build works.

david-german-tri, 2016-04-21:
@jamiesnape The AMI number you mentioned appears to be a Linux AMI, and there are no Windows AMIs in us-west-1.  Did you mean ami-18504972 (N. Virginia)?

billhoffman, 2016-04-21:
Let's see if the MP flag removal fixes it.

david-german-tri, 2016-04-21:
@billhoffman Indeed, but launching a buildslave to reproduce a failure is something TRI devs need to know how to do anyway, so I want to push forward with it.

jamiesnape, 2016-04-21:
No, I meant us-west-1 (N. California):

https://us-west-1.console.aws.amazon.com/ec2/v2/home?region=us-west-1#Images:sort=name

You cannot use us-east-1 without risking the integrity of the build farm.

jamiesnape, 2016-04-21:
Sorry, I copied the wrong AMI. One moment.

jamiesnape, 2016-04-21:
ami-50e09f30

jwnimmer-tri, 2016-04-21:
I believe this is fixed now that `/MP` or whatever was removed?

closed 2016-04-22 by david-german-tri

billhoffman, 2016-04-23:
Yes, /MP was removed and it fixed the problem, we are back to single threaded builds...