RobotLocomotion/drake #2136: VERBOSE=1 builds are broken

jwnimmer-tri, 2016-04-21:
On a clean clone of drake:
```
env PATH=/usr/lib/ccache:$PATH CXX=g++-4.9 CC=gcc-4.9 make options   # c, e, c, e, g
env PATH=/usr/lib/ccache:$PATH CXX=g++-4.9 CC=gcc-4.9 make VERBOSE=true
...
[ 41%] Performing build step for 'director'
cd /home/jwnimmer/jwnimmer-tri/drake/externals/director/distro/pods/drake-distro && make BUILD_PREFIX=/home/jwnimmer/jwnimmer-tri/drake/build BUILD_TYPE=Release
make[4]: Entering directory `/home/jwnimmer/jwnimmer-tri/drake/externals/director/distro/pods/drake-distro'
make -C pod-build true.SILENT
make: Entering an unknown directory
make: *** pod-build: No such file or directory.  Stop.
```

jamiesnape, 2016-04-21:
Director is a CMake project. You need to set `-DCMAKE_VERBOSE_MAKEFILE=ON` in CMake and then just run `make` without the `VERBOSE=true`.

jamiesnape, 2016-04-21:
See also https://github.com/RobotLocomotion/drake/blob/master/CMakeLists.txt#L386.

jwnimmer-tri, 2016-04-21:
I am merely following the instructions: http://drake002.csail.mit.edu/drake/sphinx/from_source.html

If they are wrong, then I guess this ticket turns into fixing the instructions, instead of the build system.

jamiesnape, 2016-04-21:
In general, they are wrong. They probably do not throw an error the top-level Makefile and it is hand written, nor for any Makefile PODs.

billhoffman, 2016-04-21:
Um..Not quite following all of this, but in a regular non-pod cmake based project if you run cmake, then you can run make VERBOSE=1 to get verbose output.  If it is not passing it correctly down the external project part then $(MAKE) is not being used.  Either way I don't see how make VERBOSE=1 should break a build.

jwnimmer-tri, 2016-04-21:
Look at the director Makefile.  The "true.SILENT" target is coming from the way it by-default enables `.SILENT` unless an environment variable is set.  It looks to me like that Makefile is being invoked as "make" (with the default target), instead of "make all" like it is written for.

jamiesnape, 2016-04-21:
@billhoffman I think it is a PODs issue.

jamiesnape, 2016-04-21:
Normally, CMake writes its own Makefile and either works, I would think.

billhoffman, 2016-04-21:
I see it is an issue with the Makefile here: https://github.com/RobotLocomotion/director/blob/master/distro/pods/drake-distro/Makefile

david-german-tri, 2016-04-21:
I think the "right" fix is to add `.PHONY: all` to the top of that director Makefile.   I'll send them a PR.

References:

https://cmake.org/pipermail/cmake/2007-July/015439.html (featuring @billhoffman circa 2007)

https://www.gnu.org/software/make/manual/html_node/Goals.html

david-german-tri, 2016-04-21:
Actually, I'll just move the SILENT target to the bottom.  That's semantically better, I think.

david-german-tri, 2016-04-21:
https://github.com/RobotLocomotion/director/pull/256

david-german-tri, 2016-04-22:
To resolve this issue, we need to update our Director version to https://github.com/RobotLocomotion/director/commit/cfc7702bf9278707bca82d0393bba8611513ea15 or later.  PR coming soon.

PR #2180 (2016-04-25): Update to RobotLocomotion/director cfc770 to fix VERBOSE=1 in superbuild
Closes #2136 

This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/2180)
closed 2016-04-25 by david-german-tri