RobotLocomotion/drake #2138: Reported: unable to make -j8 in superbuild

david-german-tri, 2016-04-21:
From @jwnimmer-tri:  a clean superbuild with `env PATH=/usr/lib/ccache:$PATH CXX=g++-4.9 CC=gcc-4.9 make options (ceceg) && env PATH=/usr/lib/ccache:$PATH CXX=g++-4.9 CC=gcc-4.9 make -j8` randomly dies in superbuild externals for no discernable reason.  I eventually just gave up, did it under -j1, and now just use -j8 in drake.

jamiesnape, 2016-04-21:
We run `-j8` on non-MATLAB Ubuntu CI builds.

jwnimmer-tri, 2016-04-21:
But CI doesn't run `make` like I do, does it?  It runs `cmake`?

jamiesnape, 2016-04-21:
Which runs `make -j8` under the hood.
```cmake
set(CTEST_CMAKE_GENERATOR "Unix Makefiles")
set(CTEST_BUILD_FLAGS -j8)
```

jwnimmer-tri, 2016-04-21:
Somehow, even though I don't have matlab, spotless is still trying to get involved, and of course is not working.  I am working to find a clear reproduction recipe.

jwnimmer-tri, 2016-04-21:
Here's my steps on Ubuntu 14.04 without matlab installed:
 ```
jwnimmer@call:~/tmp$ git clone https://github.com/RobotLocomotion/drake.git drake-distro
jwnimmer@call:~/tmp$ cd drake-distro/
jwnimmer@call:~/tmp/drake-distro$ env CXX=g++-4.9 CC=gcc-4.9 make options  #ceceg
Configuring with CMAKE_FLAGS: -DCMAKE_BUILD_TYPE=Release -DPOD_BUILD=TRUE
...
-- CMAKE_INSTALL_PREFIX=/home/jwnimmer/tmp/drake-distro/build
-- Found Java: /usr/bin/java (found version "1.8.0.77") 
Looked for MATLAB but could not find it.
-- Preparing to build cmake with dependencies: 
...
-- Preparing to build spotless with dependencies: 
...
-- Preparing to build drake with dependencies: cmake;eigen;googletest;lcm;bot_core_lcmtypes;libbot;bullet;swig_matlab;swigmake;yaml_cpp
...
-- Build files have been written to: /home/jwnimmer/tmp/drake-distro/pod-build
jwnimmer@call:~/tmp/drake-distro$ env CXX=g++-4.9 CC=gcc-4.9 make download-all
Configuring with CMAKE_FLAGS: -DCMAKE_BUILD_TYPE=Release -DPOD_BUILD=TRUE
-- CMAKE_INSTALL_PREFIX=/home/jwnimmer/tmp/drake-distro/build
Looked for MATLAB but could not find it.
...
-- Preparing to build spotless with dependencies: 
...
jwnimmer@call:~/tmp/drake-distro$ env CXX=g++-4.9 CC=gcc-4.9 make -j8
...
[ 70%] CMake Error at CMakeLists.txt:16 (message):
  Could not find matlab executable
...
[ 11%] make[3]: *** [spotless-prefix/src/spotless-stamp/spotless-configure] Error 1
```

jwnimmer-tri, 2016-04-21:
Possibly a regression from #2109.  Next question: why didn't CI fail?

jwnimmer-tri, 2016-04-21:
When I try the make command over and over again with `-j8`, the superbuild continues to fail.

When I remove the `-j8`, the build instead reports:
```
[ 56%] Performing update step for 'download-spotless'
[ 56%] No configure step for 'download-spotless'
[ 56%] No build step for 'download-spotless'
[ 57%] No install step for 'download-spotless'
[ 57%] Completed 'download-spotless'
[ 59%] Built target download-spotless
```
and is totally happy.

david-german-tri, 2016-04-21:
I can reproduce this on Windows too.

sherm1, 2016-04-21:
Similar symptoms reported in discussion on #2118. 

/cc @mwoehlke-kitware

billhoffman, 2016-04-21:
Must be a missing depend.  With j8 it is building in a different order.

jwnimmer-tri, 2016-04-21:
I guess?  But the fact that anything related to spotless happens at all, when there is no matlab, seems like a bigger issue.

david-german-tri, 2016-04-21:
@jwnimmer-tri I agree.  I think https://github.com/RobotLocomotion/drake/pull/2146 is the fix, PTAL?

billhoffman, 2016-04-21:
LGTM

jwnimmer-tri, 2016-04-21:
Even with #2146 PR fix for this defect, we still need to understand why the CI didn't catch this.  The (for example) most recent [linux-gcc-continuous-release](https://drake-jenkins.csail.mit.edu/view/Continuous/job/linux-gcc-continuous-release/164/consoleFull) does not mention spotless in its build log output, which means that what the CI is doing differs from what the developer instructions were doing, which leads to PRs that pass CI, but break for end-users.  Bad mojo.

jamiesnape, 2016-04-21:
CI cannot select options interactively, unfortunately.

jwnimmer-tri, 2016-04-21:
Right.  Either we teach CI to feed keystrokes into 'make options', or we stop telling developers to run 'make options'.

mwoehlke-kitware, 2016-04-21:
>  Either we teach CI to feed keystrokes into 'make options'

What would that do that can't be accomplished by running CMake twice (once with no options, once with the options to be "interactively selected")?

david-german-tri, 2016-04-21:
Before we start discussing solutions, I for one could use a little more clarity on why there was a discrepancy in the first place.  @jamiesnape How does WITH_SPOTLESS get set to false in the no-MATLAB CI builds?

jamiesnape, 2016-04-21:
A cache variable.

closed 2016-04-21 by david-german-tri

david-german-tri, 2016-04-21:
OK.  And am I correct in thinking that cache variable is derived from this line?

https://github.com/billhoffman/drake/blob/build-scripts/scripts/drake_jenkins_ctest.cmake#L360

jamiesnape, 2016-04-21:
Yes.

david-german-tri, 2016-04-21:
OK.  In that case, I claim #2146 is sufficient to resolve the discrepancy between the default `make options` build and CI: now, in both cases, when matlab is off, spotless is off.  

I also think @jwnimmer-tri is right: we need some mechanism to make sure the default superbuild config agrees with the one that `drake_jenkins_ctest.cmake` generates.  Opened https://github.com/RobotLocomotion/drake/issues/2148

PR #2146 (2016-04-21): Ensure that WITH_SPOTLESS is off if matlab is not found.
Closes #2138.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/2146) &emsp; Multiple assignees:&emsp;[image]&nbsp;<a href="/robotlocomotion/drake/pulls/assigned/jamiesnape">jamiesnape</a>,&emsp;[image]&nbsp;<a href="/robotlocomotion/drake/pulls/assigned/jwnimmer-tri">jwnimmer-tri</a>